thingsboard/thingsboard-gateway #1587: [HELP] Docs for OPCUA connector configuration

andersenthomas98, 2024-11-06:
**Describe the issue**
Hi, where can I find the documentation for the fields in the OPCUA connector configuration? The default version that comes up in thingsboard cloud looks like this: 

```json
{
  "server": {
    "url": "localhost:4840/freeopcua/server/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 3600000,
    "pollPeriodInMillis": 5000,
    "enableSubscriptions": true,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "deviceNodePattern": "Root\\.Objects\\.Device1",
      "deviceNodeSource": "path",
      "deviceInfo": {
        "deviceNameExpression": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "deviceNameExpressionSource": "path",
        "deviceProfileExpression": "Device",
        "deviceProfileExpressionSource": "constant"
      },
      "attributes": [
        {
          "key": "temperature °C",
          "type": "path",
          "value": "${ns=2;i=5}"
        }
      ],
      "timeseries": [
        {
          "key": "humidity",
          "type": "path",
          "value": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Humidity}"
        },
        {
          "key": "batteryLevel",
          "type": "path",
          "value": "${Battery\\.batteryLevel}"
        }
      ],
      "rpc_methods": [
        {
          "method": "multiply",
          "arguments": [
            {
              "type": "integer",
              "value": 2
            },
            {
              "type": "integer",
              "value": 4
            }
          ]
        }
      ],
      "attributes_updates": [
        {
          "key": "deviceName",
          "type": "path",
          "value": "Root\\.Objects\\.Device1\\.serialNumber"
        }
      ]
    }
  ],
  "logLevel": "INFO",
  "name": "test",
  "enableRemoteLogging": false,
  "id": "dc087da7-8682-45b8-a4ee-10eb5cba1591"
}
```
However, this does not look like it matches what is documented [here](https://thingsboard.io/docs/iot-gateway/config/opc-ua/): 

```json
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "localhost:4840/freeopcua/server/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions":false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "CertificateNumber",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "temperature °C",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Temperature}"
          },
          {
            "key": "batteryLevel",
            "path": "${Battery\\.batteryLevel}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [2, 4]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
```
For instance, there is a new field called `"type"` in the `"timeseries"` subsection.

imbeacon, 2024-11-07:
Hi @andersenthomas98,

Thank you for your interest in ThingsBoard IoT Gateway,
You are right, we have some changes in configuration, documentation is in progress. 
But you can ask here about parameters that you interested in.

type in timeseries is a parameter that allows UI to identify what type it should show in basic configuration mode for OPC-UA connector, it can be `constant`, `identifier`, `path`. It doesn’t important for the gateway processing, but allows to show correct type on UI. New UI fo gateways will be available in ThingsBoard 3.9.

andersenthomas98, 2024-11-07:
Ok, understand :). Can you elaborate on how to use the `deviceNodePattern`- and `deviceNameField`.? From my understanding, it seems like the `deviceNodePattern` must be a regular expression to find a node. Is this field mandatory? For my configuration I only want to define one OPCUA device, therefore, I would presume if `deviceNodePattern` does not exist the configuration would just assign all timeseries and attributes to the device named using the `deviceNameField`.

imbeacon, 2024-11-07:
deviceNodePattern, is the similar like it is defined in the current documentation, gateway uses it to find the base node for device, all path patterns for device can be relative (from device node or absolute - from the root). deviceNamePattern also can be a constant value.

`deviceNodePattern` is a base for device search, so it is mandatory, the gateway won’t be able to locate nodes without it. 
If you want to have only one device you can try to set it to Objects node, or any on the highest level and set deviceNamePattern as a constant string value.

andersenthomas98, 2024-11-07:
Thanks. Another question: What is the `pollPeriodInMillis` inside the `server` section?

imbeacon, 2024-11-08:
We added separated parameters for scanning period and polling period, earlier we did the both things at the same time, but if you know that your node hierarchy won’t change frequently - you can set a big value for `scanPeriodInMillis`, but keep low `pollPeriodInMillis`. `pollPeriodInMillis` has effect only if you have disabled subscriptions.

andersenthomas98, 2024-11-08:
Ok, thanks again! Do you have a timeline for expected release of the official docs?

andersenthomas98, 2025-01-08:
Hi again! In the gateway's general configuration section there are multiple rate limiting variables that can be set. These appear to be default: 
```
"rateLimits": "50:1,100:60",
"dpRateLimits": "50:1,100:60",
"messagesRateLimits": "50:1,100:60",
"deviceMessagesRateLimits": "50:1,100:60",
"deviceRateLimits": "50:1,100:60",
"deviceDpRateLimits": "50:1,100:60"
```

What are the differences between them?

closed 2025-09-08 by samson0v