thingsboard/thingsboard-gateway #1589: Feature proposal - Modbus batch reading

TheSamabo, 2024-11-07:
Hello
I have a feature proposal which i would like to discuss. Its about batch reading from modbus devices. I've seen this as a default implementation in multiple PLC IDEs which maps datapoints (Timeseries or attributes) to a a specific "Range" of bytes from the modbus response.
Where the gateway would read for example 16 registers starting at address 1000 (1000,1001, ... 1015) and then map datapoints to timeseries or attributes (example: "Temperature" = Float32 which would translate to value being built from registers 1000 and 1001).

This feature would require to change the configuration structure for the telemetry in the modbus connectors 
Example:
```
"timeseries": [
    {
       "starting_address": 1000,
       "register_count": 16,
       "datapoints": {
             "<KEY_NAME>":  "<DATATYPE>"
             "Temperature" : "Float32'
        }
    }
]
```
This would reduce the number of requests that have to be made against the modbus device. In othere word it would increase the throughput of data that can be read at higher frequency.  The above example is a rough idea, the datapoints each might define the number of registers that the GW should build the output value but its up to discussion. 

Thank you
Sam

ziotibia81, 2025-01-30:
This feature could be very useful. Some devices that implement Modbus RTU over TCP internally consist of a TCP-to-Serial converter connected to the device's microprocessor. In this scenario, it is important to have a delay between each request. Reading many consecutive registers in a singlerequest is more efficient with this type of device.

samson0v, 2025-09-24:
Hi @TheSamabo, @ziotibia81, @DouglasF3rreira,  thank you for your interest in ThingsBoard IoT Gateway!
We are pleased to announce that this feature will be included in the next release of Gateway.
So let's stay connected!