thingsboard/thingsboard-gateway #1591: [BUG] *Bacnet connector not found*

Isac-enerlab, 2024-11-07:
Bacnet Connector

We tried to activate version 3.6. The connector is active but does not locate the devices due to the lack of the BACnet connector. 
I installed the "bacpypes" library, with no results.
I downgraded Python to version 3.9 but it didn't give any results either.

I have units that use Modbus communication protocol that I also need to test.

{
  "thingsboard": {
    "host": "thingsboard.cloud",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600,
      "configuration": null,
      "commands": []
    }
.......

  "connectors": [
    {
      "type": "bacnet",
      "name": "Bacnet",
      "configuration": "bacnet.json"
    }
  ]
}

{
  "general": {
    "objectName": "TB_gateway",
    "address": "0.0.0.0:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1024,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "SCT_F0_CWP_EM1",
      "deviceType": "Energy Meter",
      "address": "192.168.XX.XXX:4xxx8",
      "pollPeriod": 60000,
      "timeseries": [
        {
          "key": "kWh",
          "objectId": "analogValue:1XX",
          "propertyId": "presentValue"
        }
     }
  ],
  "logLevel": "DEBUG",
  "name": "Bacnet",
  "enableRemoteLogging": false,
  "id": "xxxxxxxxxxxxxxxxxxxxxxxx",
  "configVersion": 3.6
}

Nov 07 16:36:34 GW-UB24-04-TB36 python3[28813]: BACnet library not found - installing...
Nov 07 16:36:34 GW-UB24-04-TB36 python3[28813]: 2024-11-07 16:36:34 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 73 -No module named 'asyncore'
Nov 07 16:36:34 GW-UB24-04-TB36 python3[28813]: 2024-11-07 16:36:34 - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 873 - Connector implementation not found for Bacnet
Nov 07 16:36:34 GW-UB24-04-TB36 python3[28813]: 2024-11-07 16:36:34 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 876 - The following error occurred during importing connector class: No module named 'asyncore'
Nov 07 16:36:34 GW-UB24-04-TB36 python3[28813]: 2024-11-07 16:36:34 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 876 - The following error occurred during importing connector class: No module named 'asyncore'
Nov 07 16:36:35 GW-UB24-04-TB36 python3[28813]: 2024-11-07 16:36:35 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 187 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 1.2, 'freeMemory': 11.8, 'freeDisk': '52.8G', 'gwProcessCpuUsage': 5.0, 'gwMemory': 1.5, 'gwProcessMemoryFullInfo': {'rss': 59645952, 'vms': 997888000, 'shared': 23330816, 'text': 6352896, 'lib': 0, 'data': 147378176, 'dirty': 0, 'uss': 53690368, 'pss': 54851584, 'swap': 0}, 'msgsSentToPlatform': 9, 'msgsReceivedFromPlatform': 4}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 0, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {}}

Operating System: Ubuntu 24.04.1 LTS
    Kernel: Linux 6.8.0-1010-raspi
    Architecture: arm64
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version 3.12.3

samson0v, 2024-12-03:
Hi @Isac-enerlab!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

imbeacon, 2024-12-11:
This issue closed, because relates to outdated implementation of connector, please try the latest version (3.6.2), and feel free to open a new issue if you discover some misbehaviour.

closed 2024-12-11 by imbeacon