thingsboard/thingsboard-gateway #1596: [HELP]Configure Virtual Bacnet device with Thingsboard BACnet Connector

sanjayN4497, 2024-11-13:
**Describe the issue**

Issue-1:

I'm using python BAC0 based code to setup create virtual BACnet device, this code hosted on EC2 instance ubuntu instance. Also i tried with my windows machine.

Now in thingsboard i can see below logs only unable to see device and it's objects mentioned in json and error coming in tb_gateway.py file is as below:

**site-packages\thingsboard_gateway\connectors\bacnet\bacnet_uplink_converter.py", line 53, in convert
    count=len(dict_result["attributes"]))
              ~~~~~~~~~~~^^^^^^^^^^^^^^
TypeError: string indices must be integers, not 'str'** 

as well as logs in thingsboard connector are as below:

**2024-11-13 13:20:21	DEBUG	[tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application __iam_cb - 164 - Received IAm Response: <bacpypes.apdu.ReadPropertyACK(12,1) instance at 0x29963b34050>
2024-11-13 13:20:21	DEBUG	[tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application indication - 73 - 65.0.198.181
2024-11-13 13:20:21	DEBUG	[tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application indication - 68 - Received IAmRequest from device with ID: 11111 and address xx.xx.xx.xx:47808
2024-11-13 13:20:21	DEBUG	[tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application indication - 73 - 65.0.198.181
2024-11-13 13:20:21	DEBUG	[tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application indication - 68 - Received IAmRequest from device with ID: 11111 and address xx.xx.xx.xx:47808
2024-11-13 13:20:21	DEBUG	[bacnet_connector.py] - bacnet_connector scan_network - 227 - WhoIsRequest has been sent.
2024-11-13 13:20:21	DEBUG	[bacnet_connector.py] - bacnet_connector run - 81 - WhoIsRequest has been sent.
2024-11-13 13:20:21	DEBUG	[bacnet_connector.py] - bacnet_connector scan_network - 227 - WhoIsRequest has been sent.**

also i tried with latest gatway version & 3.5.1 version but issue is remain same.

Issue-2 
on amazon ec2 ubuntu unable to add private IP here

device1 = lite(ip='xx.xx.xx.xx/24', port = '47808', deviceId = '11111') # tried with connect also but same issue

**BAC0.core.io.IOExceptions.InitializationError: Gros probleme : Error starting app: [Errno 99] Cannot assign requested address. Address requested :**

**Configuration** (Attach your configuration file)
[bacnet.txt](https://github.com/user-attachments/files/17728668/bacnet.txt)
[sample_device.txt](https://github.com/user-attachments/files/17728669/sample_device.txt)

**Connector name (If you need help with some connector/converter):**
BACnet connector

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version latest , 3.5.1
 - Python version 3.12

sanjayN4497, 2024-12-04:
Hi @samson0v ,

Thanks for update !!

Now I'm facing below error:

2024-12-04 19:50:49 - |ERROR| - [tb_logger.py] - tb_logger - exception - 161 - an error has occurred: 'str' object has no attribute 'get'
Traceback (most recent call last):
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\core.py", line 165, in run
    fn(*args, **kwargs)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\udp.py", line 301, in _response
    peer.response(pdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\udp.py", line 79, in response
    self.director.response(pdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\bvllservice.py", line 41, in confirmation
    self.multiplexer.confirmation(self, pdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\bvllservice.py", line 179, in confirmation
    self.annexJ.response(PDU(pdu, source=src, destination=dest))
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\bvllservice.py", line 312, in confirmation
    self.response(rpdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\bvllservice.py", line 402, in confirmation
    self.response(xpdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\netservice.py", line 213, in confirmation
    self.adapterSAP.process_npdu(self, npdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\netservice.py", line 573, in process_npdu
    self.response(apdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\appservice.py", line 1258, in confirmation
    tr.confirmation(apdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\appservice.py", line 410, in confirmation
    self.await_confirmation(apdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\appservice.py", line 567, in await_confirmation
    self.response(apdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\appservice.py", line 400, in response
    self.ssmSAP.sap_response(apdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\comm.py", line 537, in sap_response
    self.serviceElement.confirmation(*args,**kwargs)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\appservice.py", line 1561, in confirmation
    self.sap_response(xpdu)
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\bacpypes\comm.py", line 537, in sap_response
    self.serviceElement.confirmation(*args,**kwargs)
[... cut ...]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\sanjaynandaniya\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\thingsboard_gateway\connectors\bacnet\bacnet_uplink_converter.py", line 39, in __init__
    self.__device_report_strategy = ReportStrategyConfig(self.__config.get(REPORT_STRATEGY_PARAMETER))
                                                         ^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get'

imbeacon, 2024-12-11:
This issue closed, because relates to outdated implementation of connector, please try the latest version (3.6.2), and feel free to open a new issue if you discover some misbehaviour.

closed 2024-12-11 by imbeacon