thingsboard/thingsboard-gateway #1603: [BUG] Bacnet connector stopping - ReportStrategyConfig failling

elgutierrez, 2024-11-18:
**Describe the bug**
I'm running the gateway using Docker and the 3.6.1 version on Bacnet. The config is OK and the device responds to the IAM request. But it won't report the data to the platform because of this error where it can't access the Report Strategy

**Connector name (If bug in the some connector):**
BACnet

**Error traceback (If available):**
```
tb-gateway  | 2024-11-18 13:29:31 - |ERROR| - [tb_logger.py] - tb_logger - exception - 161 - an error has occurred:'str' object has no attribute 'get'
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/core.py", line 165, in run
tb-gateway  |     fn(*args, **kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/udp.py", line 301, in _response
tb-gateway  |     peer.response(pdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/udp.py", line 79, in response
tb-gateway  |     self.director.response(pdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
tb-gateway  |     self.serverPeer.confirmation(*args, **kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 41, in confirmation
tb-gateway  |     self.multiplexer.confirmation(self, pdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 179, in confirmation
tb-gateway  |     self.annexJ.response(PDU(pdu, source=src, destination=dest))
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
tb-gateway  |     self.serverPeer.confirmation(*args, **kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 312, in confirmation
tb-gateway  |     self.response(rpdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
tb-gateway  |     self.serverPeer.confirmation(*args, **kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 402, in confirmation
tb-gateway  |     self.response(xpdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
tb-gateway  |     self.serverPeer.confirmation(*args, **kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/netservice.py", line 213, in confirmation
tb-gateway  |     self.adapterSAP.process_npdu(self, npdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/netservice.py", line 573, in process_npdu
tb-gateway  |     self.response(apdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
tb-gateway  |     self.serverPeer.confirmation(*args, **kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/appservice.py", line 1258, in confirmation
tb-gateway  |     tr.confirmation(apdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/appservice.py", line 410, in confirmation
tb-gateway  |     self.await_confirmation(apdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/appservice.py", line 567, in await_confirmation
tb-gateway  |     self.response(apdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/appservice.py", line 400, in response
tb-gateway  |     self.ssmSAP.sap_response(apdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 537, in sap_response
tb-gateway  |     self.serviceElement.confirmation(*args,**kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/appservice.py", line 1561, in confirmation
tb-gateway  |     self.sap_response(xpdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 537, in sap_response
tb-gateway  |     self.serviceElement.confirmation(*args,**kwargs)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/app.py", line 491, in confirmation
tb-gateway  |     self._app_complete(apdu.pduSource, apdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/app.py", line 454, in _app_complete
tb-gateway  |     queue.complete_io(queue.active_iocb, apdu)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/iocb.py", line 753, in complete_io
tb-gateway  |     IOController.complete_io(self, iocb, msg)
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/iocb.py", line 623, in complete_io
tb-gateway  |     iocb.trigger()
tb-gateway  |   File "/root/.local/lib/python3.11/site-packages/bacpypes/iocb.py", line 172, in trigger
tb-gateway  |     fn(self, *args, **kwargs)
tb-gateway  |   File "/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 167, in __iam_cb
tb-gateway  |     value = self.__connector.default_converters["uplink_converter"]("{}", self._log).convert(None, apdu)
tb-gateway  |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
tb-gateway  |   File "/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py", line 39, in __init__
tb-gateway  |     self.__device_report_strategy = ReportStrategyConfig(self.__config.get(REPORT_STRATEGY_PARAMETER))
tb-gateway  |                                                          ^^^^^^^^^^^^^^^^^
tb-gateway  | AttributeError: 'str' object has no attribute 'get'
```

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04 + Docker
 - Thingsboard IoT Gateway version: 3.6.1

elgutierrez, 2024-11-18:
I've been investigating and the issue seems to be coming from this like:

```
value = self.__connector.default_converters["uplink_converter"]("{}", self._log).convert(None, apdu)
```
So the `config` param is a string. If I change it to an empty object, at least this issue disappears, but then there's a new error:

```
  File "thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py", line 52, in convert
    device_name=self.__config.get("deviceName", config[1].get("name", "BACnet device")),
                                                ~~~~~~^^^
TypeError: 'NoneType' object is not subscriptable
```

Which is pretty normal, as the `convert` method is called with `config = None` 

```
    def convert(self, config, data):
        converted_data = ConvertedData(
            device_name=self.__config.get("deviceName", config[1].get("name", "BACnet device")),
            device_type=self.__config.get("deviceType", "default")
        )
```

I appreciate the effort and really like the Thingsboard platform, so much so that I use it in production in my business and I'm a proud subscriber of the hosted TB Cloud. But I'd like to mention that on every release, some bugs are fixed but then 5 other new bugs appear. The BACnet connector is broken since the version 3.5.2, which is the last version that I managed to make work. But I'd love to upgrade to a newer version so I can use the SQLite storage ([which is broken on 3.5.1](https://github.com/thingsboard/thingsboard-gateway/issues/1446)) but I can't and I'm stuck in this older version.

I'd love to help but without proper typing or tests, it's hard to get all the codebase context.

samson0v, 2024-12-03:
Hi @elgutierrez!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

elgutierrez, 2024-12-05:
Hey @samson0v , thanks for the update. I'll see when I have a slot for testing that. Is there an ETA for the new release?

imbeacon, 2024-12-11:
Hi @elgutierrez,

The new release is available, feel free to try it. I’m closing this issue, because it relates to deprecated for now connector. But, please open a new, if you discover some issues with new BACnet connector.

closed 2024-12-11 by imbeacon