thingsboard/thingsboard-gateway #1613: need Help modbus(_serial).json config

pro100805, 2024-12-08:
Good afternoon, help me to understand for interaction with the device I use 16int format.
As far as I understand the following settings apply to it, for example
        "timeseries": [
          {
            "tag": "voltage",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 8
          }
         ]
and for RPC 
        "rpc": [
          {
            "tag": "setSet_Point",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 17
          },
          {
            "tag": "getSet_Point",
            "type": "16int",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 17
          }
         ]
the question is whether everything is correct 
and the second question is there a difference in the configuration depending on what type of connection (tcp/ip or seria).

pro100805, 2024-12-08:
will this config work the same for tcp/ip and serial
        "timeseries": [
          {
            "tag": "voltage",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 8
          },
          {
            "tag": "voltage3",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 10
          },
          {
            "tag": "DO_1",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 2
          },
          {
            "tag": "DO_2",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 3
          },
          {
            "tag": "DO_3",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 4
          }
        ],
        "attributeUpdates": [],
        "rpc": [
          {
            "tag": "setSet_Point",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 17
          },
          {
            "tag": "getSet_Point",
            "type": "16int",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 17
          },
          {
            "tag": "setFAN",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 11
          },
          {
            "tag": "getFAN",
            "type": "16int",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 11
          },
          {
            "tag": "setDO_1",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 2
          },
          {
            "tag": "getDO_1",
            "type": "16int",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 2
          },
          {
            "tag": "setDO_2",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 3
          },
          {
            "tag": "getDO_2",
            "type": "16int",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 3
          },
          {
            "tag": "setDO_3",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 4
          },
          {
            "tag": "getDO_3",
            "type": "16int",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 4
          }
        ]
      }

imbeacon, 2024-12-11:
Hi @pro100805,

Your configuration looks good at first glance, I don’t see reasons why it should not work.
Due to your second question - no, there is no difference for sections `timeseries`, `attributes`, `attributeUpdates`, `rpc` for different types of connection, difference is only in connection configuration.

pro100805, 2024-12-11:
thank you for your quick answer
I am using TB 3.8.1 and GW 3.6.1 connection via RS 485 (serial port) 
In TB I have two different dashboards
one only timeseries
the second one timeseries + rpc
the problem is that 
if you use only timeseries everything works stably
but when you start using rpc you get delays in connection and the following messages appear

 thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-12-08 10:17:27 UTC; 3 days ago
   Main PID: 953 (python3)
      Tasks: 31 (limit: 2225)
     Memory: 473.1M
        CPU: 15h 25min 21.513s
     CGroup: /system.slice/thingsboard-gateway.service
             └─953 /var/lib/thingsboard_gateway/venv/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 626 - Reading failed for device Humidity_DHT function code 4 address 13 unit id 1
Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 628 - Reading failed with exception:
Dec 11 15:10:45 test-22-04-4 python3[953]: pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |INFO| - [modbus_connector.py] - modbus_connector - __function_to_device - 629 - Trying to reconnect to device Humidity_DHT
Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 626 - Reading failed for device TestDevise function code 6 address 4 unit id 1
Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 628 - Reading failed with exception:
Dec 11 15:10:45 test-22-04-4 python3[953]: pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |INFO| - [modbus_connector.py] - modbus_connector - __function_to_device - 629 - Trying to reconnect to device TestDevise
Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |INFO| - [modbus_connector.py] - modbus_connector - __function_to_device - 634 - Reconnected to device Humidity_DHT
Dec 11 15:10:45 test-22-04-4 python3[953]: 2024-12-11 15:10:45 - |INFO| - [modbus_connector.py] - modbus_connector - __function_to_device - 634 - Reconnected to device TestDevise

pro100805, 2024-12-11:
it is enough to restart the GW service and everything works stable again until we execute rpc requests (commands).