thingsboard/thingsboard-gateway #1615: [BUG] Database connection seems to be closed at the wrong moment

dwaxweiler, 2024-12-10:
**Describe the bug**
A connector fails because changes cannot be committed to the temporary SQLite storage:

**Connector name (If bug in the some connector):**
FTP Connector

**Error traceback (If available):**
```
tb-gateway  | 2024-12-10 11:00:48 - |DEBUG| - [database_connector.py] - database_connector - execute - 60 - Failed to execute changes to database
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/storage/sqlite/database_connector.py", line 58, in execute
tb-gateway  |     return self.connection.execute(*args)
tb-gateway  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
tb-gateway  | sqlite3.ProgrammingError: Cannot operate on a closed database.
tb-gateway  | 2024-12-10 11:00:48 - |DEBUG| - [database_connector.py] - database_connector - commit - 43 - Committing changes to DB
tb-gateway  | 2024-12-10 11:00:48 - |ERROR| - [tb_logger.py] - tb_logger - exception - 161 - Failed to commit changes to database
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/storage/sqlite/database_connector.py", line 46, in commit
tb-gateway  |     self.connection.commit()
tb-gateway  | sqlite3.ProgrammingError: Cannot operate on a closed database.
```

**Versions (please complete the following information):**
 - OS: Debian 12
 - Thingsboard IoT Gateway version: 3.6.1 Docker

imbeacon, 2024-12-11:
Hi @dwaxweiler,

Thank you for your interest in ThingsBoard IoT Gateway. We heard about this bug, but, unfortunately were unable to reproduce, could you provide more information, about situation when the issue appeared, like issues with internet communication/ issues with disk, maybe space, working time, etc.

imbeacon, 2025-01-10:
We have added a fix for this issue in 3.6.3, please try it and let us know if the issue still appears. Feel free to open a new issue.

closed 2025-01-10 by imbeacon