thingsboard/thingsboard-gateway #1620: [BUG] New release Tb_gateway 3.6.2

Iskvaz, 2024-12-11:
**Describe the bug**

Testing the new release with bacnet connector, The device discovery process is running, but the gateway is unable to associate the response with the expected device.

**Connector name (If bug in the some connector):**
[BACnet Connector]

**Error traceback (If available):**
 |INFO| - [tb_gateway_service.py] - tb_gateway_service- __init__ - 308 - Gateway started.
 |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 242 - Configuration update request received.
Dec 11 16:48:07 TB3-6-2UBV262 python3[51036]: 2024-12-11 16:48:07 - |INFO| - [statistics_service.py] - statistics_service- __send_statistics - 208 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 3.8, 'diskUsage': 5.9, 'freeMemory': 11.6, 'freeDisk': '52.3G', 'gwProcessCpuUsage': 15.0, 'gwMemory': 1.7, 'gwProcessMemoryFullInfo': {'rss': 67842048, 'vms': 1159168000, 'shared': 23724032, 'text': 6348800, 'lib': 0, 'data': 172797952, 'dirty': 0, 'uss': 55005184, 'pss': 59233280, 'swap': 0}, 'msgsSentToPlatform': 5, 'msgsReceivedFromPlatform': 1}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 0, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {}}
 |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 242 - Configuration update request received.
 |INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 152 - Discovering devices...
 |INFO| - [bacnet_connector.py] - bacnet_connector - indication_callback - 121 - Received APDU, from 192.168.15.210, trying to find device...
 |INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 152 - Discovering devices...
 |INFO| - [bacnet_connector.py] - bacnet_connector - indication_callback - 121 - Received APDU, from 192.168.15.210, trying to find device...
 |INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 152 - Discovering devices...

'deviceName'
"devices": [
    {
      "deviceInfo": {
        "deviceNameExpressionSource": "constant",
        "deviceNameExpression": "SCT_XX_CWP_XXX",
        "deviceProfileExpressionSource": "constant",
        "deviceProfileExpression": "Energy Meter"
      },
      "address": "192.168.XX.XXX:47808",
      "pollPeriod": 60000,
      "timeseries": [
        {
          "key": "kWh",
          "objectType": "analogValue",
          "objectId": "107",
          "propertyId": "presentValue"
        }
      ]
    }
  ]

- The gateway is sending Who-Is packets, and receiving a response.

~$ sudo tcpdump -i eth0 udp port 47808
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
16:55:34.905854 IP TB3-6-2UBV262.47808 > 192.168.15.210.47808: UDP, length 8
16:55:34.909214 IP 192.168.15.210.47808 > TB3-6-2UBV262.47808: UDP, length 20
16:56:04.915665 IP TB3-6-2UBV262.47808 > 192.168.15.210.47808: UDP, length 8
16:56:04.918534 IP 192.168.15.210.47808 > TB3-6-2UBV262.47808: UDP, length 20
^C
4 packets captured
4 packets received by filter
0 packets dropped by kernel

- The output I have after this step by step is below:

|INFO| - [bacnet_connector.py] - bacnet_connector - indication_callback - 121 - Received APDU, from 192.168.15.210, trying to find device...
|INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 152 - Discovering devices...
|INFO| - [bacnet_connector.py] - bacnet_connector - indication_callback - 121 - Received APDU, from 192.168.15.210, trying to find device...

**Versions (please complete the following information):**
 - OS: Ubuntu 24.04.1 LTS (GNU/Linux 6.8.0-1015-raspi aarch64)
 - Thingsboard IoT Gateway version 3.6.2
 - Python version 3.7

imbeacon, 2024-12-12:
Hi @Iskvaz,

Thank you for your interest in ThingsBoard IoT Gateway, could you try to use 192.168.15.210 instead of address and port for device? Also please try to set parameter `altResponsesAddresses` on the same level with device address to 192.168.15.210:47808, like:

`”altResponsesAddresses”: [“192.168.15.210:47808”]`

At the moment our goal is to receive response and correctly process it, then we will be able to see why it is unable to handle it using your first configuration.
@samson0v Please assist with further investigation.

samson0v, 2024-12-16:
@Iskvaz any updates?

Isac-enerlab, 2025-01-06:
Hello,
We did the test and it is working correctly with MODbus.
When we run BACnet, despite applying the suggestions, the device created in BACnet remains inactive.

Following logs:

|INFO| - [bacnet_connector.py] - bacnet_connector - indication_callback - 121 - Received APDU, from 192.168.15.210, trying to find device...
|WARNING| - [device.py] - device - __init__ - 47 - Device name is not provided in IAmRequest. Device Id will be used as "objectName
|ERROR| - [bacnet_connector.py] - bacnet_connector - indication_callback - 138 - Error processing indication callback: 'deviceNameExpression'
|INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 152 - Discovering devices..

How is my file configured.

  "devices": [
    {
      "address": "192.168.15.210",
      "altResponsesAddresses": [
        "192.168.15.210:47808"
      ],
      "deviceName": "XXX_XX_CWP_EM1",
      "deviceType": "Energy Meter",
      "attributes": [
        {
          "tag": "kWh",
          "type": "32float",
          "objectId": "analogValue,107"
        }
      ]
   }
]

imbeacon, 2025-01-07:
Hi @Isac-enerlab,

It looks like you are using mix of configurations, please try to use configuration like the following(with deviceInfo object):

```
  "devices": [
    {
      "address": "192.168.15.210",
      "altResponsesAddresses": [
        "192.168.15.210:47808"
      ],
      "deviceInfo": {
        "deviceNameExpressionSource": "constant",
        "deviceNameExpression": "XXX_XX_CWP_EM1",
        "deviceProfileExpressionSource": "constant",
        "deviceProfileExpression": "Energy Meter"
      },
      "attributes": [
        {
          "tag": "kWh",
          "type": "32float",
          "objectId": "analogValue,107"
        }
      ]
    }
  ]
```

It is a new configuration for BACnet, for gateway 3.6.2+ we are in progress of updating documentation.