thingsboard/thingsboard-gateway #1624: [BUG] version 3.6.2

pro100805, 2024-12-17:
After configuring the gateway and adding the connector, nothing happens for some time. The devices described in the connector do not appear on the TB server.

some time later, the following message appears

Dec 17 13:09:24 tesssst-24-04 python3[1822]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 204, in __add_slaves
Dec 17 13:09:24 tesssst-24-04 python3[1822]:     self.__add_slave(slave_config)
Dec 17 13:09:24 tesssst-24-04 python3[1822]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 195, in __add_slave
Dec 17 13:09:24 tesssst-24-04 python3[1822]:     slave = Slave(self, self.__log, slave_config)
Dec 17 13:09:24 tesssst-24-04 python3[1822]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Dec 17 13:09:24 tesssst-24-04 python3[1822]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/slave.py", line 39, in __init__
Dec 17 13:09:24 tesssst-24-04 python3[1822]:     config['host']) + ":" + str(config['port']) + ' ' + config['deviceName']
Dec 17 13:09:24 tesssst-24-04 python3[1822]:     ~~~~~~^^^^^^^^
Dec 17 13:09:24 tesssst-24-04 python3[1822]: KeyError: 'host'
Dec 17 13:09:24 tesssst-24-04 python3[1822]: 2024-12-17 13:09:24 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __connect_with_connectors - 1022 - Cannot find separated report strategy for connector 'Serial'.   The main report strategy     will be used as a connector report strategy.

 - OS:  Ubuntu 24.04
 - Thingsboard IoT Gateway version 3.6.2
 - Python version 3.12.3

samson0v, 2024-12-18:
Hi, @pro100805, thanks for your interest in ThingsBoard IoT Gateway!
We will fix this bug as soon as possible.

samson0v, 2024-12-18:
@pro100805 fix is available in the master branch. If something went wrong, let us know.

pro100805, 2024-12-18:
Thank you for your prompt reply. 
I am using DEB package for my installation. Please give instructions how to integrate your modified MASTER BRANCH into my installation. If possible, thank you in advance

tangchengxiang666, 2024-12-18:
> @pro100805修复已在 master 分支中提供。如果出现问题，请告知我们。

After modification according to the fix file, __get_master in modbus_connector.py file will show slave. host as None.  Is this an error?
[image]

pro100805, 2024-12-18:
Thank you, you've given me an idea and I think these are the changes that need to be made.
[image]

pro100805, 2024-12-18:
I'll make the changes and let you know what the result is.

samson0v, 2024-12-18:
@tangchengxiang666, it is ok

pro100805, 2024-12-18:
It's not as easy as I thought. It's possible that you need to rebuild something after the changes. I don't think I can do it without help.

samson0v, 2024-12-18:
@pro100805 deb file with the latest changes:

[python3-thingsboard-gateway.deb.zip](https://github.com/user-attachments/files/18178346/python3-thingsboard-gateway.deb.zip)

pro100805, 2024-12-18:
The devices described in the connector do not appear on the TB server.

the following message appears

Dec 18 11:11:34 tesssst-24-04 python3[1005]:   File "/usr/lib/python3.12/asyncio/events.py", line 88, in _run
Dec 18 11:11:34 tesssst-24-04 python3[1005]:     self._context.run(self._callback, *self._args)
Dec 18 11:11:34 tesssst-24-04 python3[1005]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 223, in __process_requests
Dec 18 11:11:34 tesssst-24-04 python3[1005]:     await self.__poll_device(slave)
Dec 18 11:11:34 tesssst-24-04 python3[1005]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 254, in __poll_device
Dec 18 11:11:34 tesssst-24-04 python3[1005]:     self.__log.error('Failed to poll %s device: %s', slave, exc_info=e)
Dec 18 11:11:34 tesssst-24-04 python3[1005]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/tb_utility/tb_logger.py", line 158, in error
Dec 18 11:11:34 tesssst-24-04 python3[1005]:     super(TbLogger, self).error(msg, *args, **kwargs)
Dec 18 11:11:34 tesssst-24-04 python3[1005]: Message: 'Failed to poll %s device: %s'
Dec 18 11:11:34 tesssst-24-04 python3[1005]: Arguments: (<Slave(Modbus slave processor for unit 1 on port /dev/ttyUSB0 Reboot, started daemon 139179801446080)>,)

tangchengxiang666, 2024-12-19:
> @tangchengxiang666，沒有關係

After repairing it using the file you provided, it is still not available for serial cases

imbeacon, 2024-12-19:
@samson0v Please assist.

closed 2025-04-11 by imbeacon