thingsboard/thingsboard-gateway #1631: [HELP] The TCP socket connector doesnt process the data in IoT Gateway

cemakpolat, 2024-12-22:
**Describe the issue**

My purpose is to bind a TCP device that generate random device to the thingsboard iot gateway and then visualize the data. I try to connect a local python tcp client to the thingsboard iot gateway socket connector running in Docker. In the logs, i see that the connection is established, however, afterwards the transmitted data doesnt appear and any TCP related device is created in thingsboard dashboard. IoT gateway runs in a docker container and the python code runs on the host.

I am not sure whether I understood enough the socket connector,and it might be the case that i configured wrongly because of the lack of understandign. I dont understand the relation between devices array and general socket address. My expection is to connect to iot gateway and push the data to the gateway, however, i dont know how the device data is figured out by the iot gateway itself.

One another thing that I cannot comprehend is the description in this page, https://thingsboard.io/docs/iot-gateway/config/socket/, says:
"The purpose of this Connector is to connect to your server using TCP or UDP connection type. This Connector is useful when you have a local server in your facility or corporate network and want to push data from the server to ThingsBoard."

This means, I should create a socket server, and device based on the device description in the iot gateway will connect to my server and then i would push the data. When I look at this code https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/connectors/socket/socket_connector.py, I see something different.

It would be better if you can explain the logic of the code and working mechanism of this connector. You may also offer the solution if you see at the first glance. Thanks for your effort and happy new year!

**Configuration** 
Here is the simple JSON configuration in thingsboard dashboard, it accepts the data from all interfaces, otherwise, i cannot establish the connection between python client and thingsboard iot gateway connector.

{
  "socket": {
    "type": "TCP",
    "address": "0.0.0.0",
    "port": 50000,
    "bufferSize": 1024
  },
  "devices": [
    {
      "address": "127.0.0.1:50001",
      "deviceName": "DeviceTCP1",
      "deviceType": "default",
      "encoding": "utf-8",
      "telemetry": [
        {
          "key": "temp",
          "byteFrom": 0,
          "byteTo": -1
        },
        {
          "key": "hum",
          "byteFrom": 0,
          "byteTo": 2
        }
      ],
      "attributes": []
    }
  ],
  "name": "tcptest",
  "enableRemoteLogging": true,
  "logLevel": "TRACE",
  "configVersion": "3.6.2",
  "id": "46ce58c3-d4b1-4c2b-a696-f95ea33366dc"
}

**Connector name (If you need help with some connector/converter):**
Socket Connector

**Error traceback (If it was raised):**
There is no error on both side, this is the only thing that i see on the iot gateway console outputs
```
tb-gateway  | 2024-12-22 17:34:54 - |DEBUG| - [socket_connector.py] - socket_connector - run - 212 - New connection ('192.168.32.1', 59418) established
tb-gateway  | 2024-12-22 17:35:24 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 208 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 0.0, 'diskUsage': 35.9, 'freeMemory': 51.9, 'freeDisk': '59.6G', 'gwProcessCpuUsage': 0.0, 'gwMemory': 1.1, 'gwProcessMemoryFullInfo': {'rss': 69328896, 'vms': 1235976192, 'shared': 23199744, 'text': 4096, 'lib': 0, 'data': 182829056, 'dirty': 0, 'uss': 68186112, 'pss': 68818944, 'swap': 0}, 'msgsSentToPlatform': 3, 'msgsReceivedFromPlatform': 0}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 0, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {}}
```
Here my the client side console output
```
DeviceTCP1: Connecting to 127.0.0.1:50000...
DeviceTCP1: Sent telemetry -> b'\x00\x02\x000'
DeviceTCP1: Sent telemetry -> b'\x00C\x00\x06'
DeviceTCP1: Sent telemetry -> b'\x002\x009'
```
** Python Client** 
```
class TCPClient:
    def __init__(self, server_host, server_port, device_name):
        self.server_host = server_host
        self.server_port = server_port
        self.device_name = device_name
        self.encoding = "utf-8"

    def generate_telemetry(self):
        """
        Generate telemetry data as binary:
        - temp: Random temperature value (0-100), 2 bytes.
        - hum: Random humidity value (0-100), 2 bytes.
        """
        temp = random.randint(0, 100)
        hum = random.randint(0, 100)
        temp_bytes = temp.to_bytes(2, byteorder='big')
        hum_bytes = hum.to_bytes(2, byteorder='big')
        # Ensure temp and hum are sequential in the buffer
        return temp_bytes + hum_bytes

    def send_data(self):
        print(f"{self.device_name}: Connecting to {self.server_host}:{self.server_port}...")
        with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as client_socket:
            try:
                client_socket.connect((self.server_host, self.server_port))
                while True:
                    telemetry_data = self.generate_telemetry()
                    client_socket.sendall(telemetry_data)
                    # Decode and log the sent data
                    # decoded_temp = telemetry_data[:4].strip(b'\x00').decode(self.encoding)
                    # decoded_hum = telemetry_data[4:].strip(b'\x00').decode(self.encoding)
                    #print(f"{self.device_name}: Sent telemetry -> temp={decoded_temp}, hum={decoded_hum}")

                    print(f"{self.device_name}: Sent telemetry -> {telemetry_data}")
                    time.sleep(2)  # Simulate periodic data sending
            except ConnectionRefusedError:
                print(f"{self.device_name}: Could not connect to server.")
            except KeyboardInterrupt:
                print(f"{self.device_name}: Stopping client.")
            except Exception as error:
                print(f"{error}: Stopping client.")

def main():
    # Client Configuration
    devices = [
        TCPClient(server_host="127.0.0.1", server_port=50000, device_name="DeviceTCP1"),
    ]

    # Start clients in separate threads
    client_threads = []
    for device in devices:
        client_thread = threading.Thread(target=device.send_data, daemon=True)
        client_threads.append(client_thread)
        client_thread.start()

    # Keep the main thread running
    try:
        while True:
            time.sleep(1)
    except KeyboardInterrupt:
        print("\nExiting program...")

if __name__ == "__main__":
    main()

```
**Versions (please complete the following information):**
 - OS: Mac OS M2, 15.2 (24C101)
 - Thingsboard IoT Gateway version: Latest
 - Python version: Python 3.11.11

cemakpolat, 2024-12-22:
ok, i think finally i got the point. In the device description address:port is just here to figure out whether the data is sent from the expected address:port number. This means, i first connect to the iot gw with the generic address:port number, and then send the data, and finally, iot gw checks whether the transmitted data comes from the address:port written in the device description. In other words address:port is used just for the filtering and forwarding the data to the right device. Therefore the help is not needed, and the ticket can be closed.

imbeacon, 2024-12-23:
Hi @cemakpolat,

Thenk you for your interest in ThingsBoard IoT Gateway.

You are right, but technically, you also can use regular expression to determine address. e.g. `127.0.0.1:*`

closed 2024-12-23 by imbeacon

cemakpolat, 2024-12-23:
Yes, this is what i did. It could understand only this usage after i checked the code :). If you have multiple device from the same IP addresses, what kind of regular expression should be used. IoT gw assings a random port number to my python client, and if want to use a second client from the same machine, i cannot use the same pattern. Do you have any other mechanism to solve this?