thingsboard/thingsboard-gateway #1634: [BUG] TLS/SSL connection has been closed

tunahanarslankerberos, 2024-12-24:
**Describe the bug**
After moving the thingsboard iot gateway from version 3.5.2 to version 3.6.2, when I tried to run it, it gave the error "TLS/SSL connection has been closed (EOF)". I do not use tls or ssl in mqtt connection, then when I moved to version 3.6.1, this problem did not occur.

**Connector name:**
MQTT connection with gateway
```
tb_gateway.json
{
  "thingsboard": {
    "host": "srv-ubuntu24",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": false,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600,
      "configuration": null,
      "commands": []
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 1,
    "handleDeviceRenaming": true,
    "security": {
      "accessToken": "ACCESS_TOKEN"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
    {
      "useGRPC": false,
      "type": "modbus",
      "name": "MODBUS",
      "configuration": "modbus.json"
    }
  ]
}
```

**Error traceback:**
2024-12-24 11:41:00 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 187 - Gateway starting...
2024-12-24 11:41:01 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 198 - ThingsBoard IoT gateway version: 3.6.2
2024-12-24 11:41:01 - |INFO| - [tb_loader.py] - tb_loader - import_module - 72 - Import AsyncModbusConnector from /home/raspberrypi/venv/lib/python3.11/site-packages/thingsboard_gateway/connectors/modbus.
2024-12-24 11:41:01 - |INFO| - [modbus_connector.py] - modbus_connector - __init__ - 88 - Starting Modbus Connector...
2024-12-24 11:41:01 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __connect_with_connectors - 1022 - Cannot find separated report strategy for connector 'MODBUS'.                                                          The main report strategy                                                          will be used as a connector report strategy.
2024-12-24 11:41:01 - |INFO| - [slave.py] - slave - connect - 166 - Connected to LAMP2
2024-12-24 11:41:01 - |INFO| - [tb_client.py] - tb_client - connect - 370 - Connecting to ThingsBoard...
2024-12-24 11:41:01 - |INFO| - [tb_client.py] - tb_client - __send_connect - 397 - Sending connect to srv-ubuntu24:1883, tls: False...
2024-12-24 11:41:02 - |ERROR| - [tb_client.py] - tb_client - connect - 390 - Error in connect: %s
Traceback (most recent call last):
  File "/home/raspberrypi/venv/lib/python3.11/site-packages/thingsboard_gateway/gateway/tb_client.py", line 374, in connect
    self.__send_connect()
  File "/home/raspberrypi/venv/lib/python3.11/site-packages/thingsboard_gateway/gateway/tb_client.py", line 398, in __send_connect
    self.client.connect(keepalive=self.__config.get("keep_alive", 120),
  File "/home/raspberrypi/venv/lib/python3.11/site-packages/tb_device_mqtt.py", line 489, in connect
    self._client.connect(self.__host, self.__port, keepalive=keepalive)
  File "/home/raspberrypi/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1435, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/home/raspberrypi/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1598, in reconnect
    self._sock = self._create_socket()
                 ^^^^^^^^^^^^^^^^^^^^^
  File "/home/raspberrypi/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 4612, in _create_socket
    sock = self._ssl_wrap_socket(sock)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/raspberrypi/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 4671, in _ssl_wrap_socket
    ssl_sock.do_handshake()
  File "/usr/lib/python3.11/ssl.py", line 1379, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLZeroReturnError: TLS/SSL connection has been closed (EOF) (_ssl.c:992)

**Versions:**
 - OS: 
      PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
      NAME="Raspbian GNU/Linux"
      VERSION_ID="12"
      VERSION="12 (bookworm)"
      VERSION_CODENAME=bookworm
      ID=raspbian
      ID_LIKE=debian
 - Thingsboard IoT Gateway version: 3.6.2
 - Python version: 3.11.2

tGallimberti-Kerberos, 2025-02-05:
Problem seem to be related to paho library

imbeacon, 2025-02-06:
Hi,

Thank you for your interest in ThingsBoard IoT Gateway. We have added handling for this error, it is available in master branch and release 3.7.0.

Please feel free to open new issue if you discover some misbehaviour.

closed 2025-02-06 by imbeacon