thingsboard/thingsboard-gateway #1635: [HELP]2024-12-25 10:55:40 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _wait_for_rate_limit_released - 749 - Rate limit reached for 1 seconds, waiting for rate limit to be released...

DuXiaoChuang, 2024-12-25:
2024-12-25 10:55:40 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _wait_for_rate_limit_released - 749 - Rate limit reached for 1 seconds, waiting for rate limit to be released...

How should this problem be solved?

imbeacon, 2025-01-06:
Hi @DuXiaoChuang,

It is a warning, it may appear during initial configuration retrieving process, before the gateway received rate limits from server. If it appears only on starting - it is fine, but if it appears continuously - please provide more information about your environment, more logs, maybe configurations and how much data you try to send to server and how often.

DuXiaoChuang, 2025-01-07:
> Hi @DuXiaoChuang,
>
> It is a warning, it may appear during initial configuration retrieving process, before the gateway received rate limits from server. If it appears only on starting - it is fine, [...]

It happens once a second, and here's what happens

{
	'deviceName': '沈阳机床T5-500Q',
	'deviceType': 'default',
	'telemetry': [{
		'刀组数': 0
	},
	{
		'每组刀具数': 0
	},
	{
		'当前刀组号': 0
	},
	{
		'当前刀具索引号': 0
	},
	{
		'总报警数': 5
	},
	{
		'报警类型': 0
	},
	{
		'报警号': 0
	},
	{
		'最后一条报警记录的编号': -16669
	},
	{
		'报警内容': '\x00\x00\x00\x00\x00\x00\x00\x00'
	},
	{
		'报警时间': 0
	},
	{
		'绝对坐标X': -1.3046000003814697
	},
	{
		'绝对坐标Z': 0.8700000047683716
	},
	{
		'绝对坐标A': 0.0
	},
	{
		'绝对坐标B': -0.0009500000160187483
	},
	{
		'绝对坐标C': 0.0
	},
	{
		'相对坐标值': -1.912109375
	},
	{
		'进给实际速度': 0.0
	},
	{
		'进给倍率': 100.0
	},
	{
		'机械坐标X': -1.3178000450134277
	},
	{
		'机械坐标Z': 0.9014999866485596
	},
	{
		'机械坐标A': 0.0
	},
	{
		'机械坐标B': -0.0009500000160187483
	},
	{
		'机械坐标C': 0.0
	},
	{
		'当前刀具号': 6
	},
	{
		'当前刀偏号': 10
	},
	{
		'加工件数': 1249
	},
	{
		'运行时间': 2272
	},
	{
		'切削时间': 0
	},
	{
		'进给编程速度': 100.0
	},
	{
		'主轴倍率': 100.0
	},
	{
		'快速倍率': 100.0
	},
	{
		'手动倍率': 100.0
	},
	{
		'手轮倍率': 0.0
	},
	{
		'CNC工作方式': 4
	},
	{
		'CNC工作状态': 0
	},
	{
		'当前运行的程序号': 0
	},
	{
		'当前程序段号': 0
	},
	{
		'手动速度': 0.0
	},
	{
		'诊断号': 0
	},
	{
		'诊断轴号': 0
	},
	{
		'诊断值': 0
	},
	{
		'诊断值类型': 1
	}],
	'attributes': [{
		'CNC型号': 'K988TA\x00\x00\x00\x00'
	},
	{
		'有效轴数': 5
	},
	{
		'上电次数': 692
	},
	{
		'机床编号': '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
	},
	{
		'轴伺服信息': 'GR'
	},
	{
		'主轴伺服信息': 'GR'
	}]
}

DuXiaoChuang, 2025-01-07:
> Hi @DuXiaoChuang,
>
> It is a warning, it may appear during initial configuration retrieving process, before the gateway received rate limits from server. If it appears only on starting - it is fine, [...]

As soon as I set the **sendDataOnlyOnChange parameter to false**, the alarm _wait_for_rate_limit_released -     appears

{
  "master": {
    "slaves": [
      {
        "host": "192.168.1.4",
        "port": 502,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "LITTLE",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 1000,
        "unitId": 1,
        "deviceName": "\u6c88\u9633\u673a\u5e8aT5-500Q",
        "sendDataOnlyOnChange": **false**,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [
          {
            "tag": "CNC\u578b\u53f7",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 5,
            "address": 63001
          },
          {
            "tag": "\u6709\u6548\u8f74\u6570",
            "type": "16int",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 63013
          },
          {
            "tag": "\u4e0a\u7535\u6b21\u6570",
            "type": "32int",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 63030
          },
          {
            "tag": "\u673a\u5e8a\u7f16\u53f7",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 5,
            "address": 63060
          },
          {
            "tag": "\u8f74\u4f3a\u670d\u4fe1\u606f",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 63100
          },
          {
            "tag": "\u4e3b\u8f74\u4f3a\u670d\u4fe1\u606f",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 63500
          }
[... 332 lines cut ...]
            "address": 31
          },
          {
            "tag": "setCPUFanSpeed",
            "type": "32int",
            "functionCode": 16,
            "objectsCount": 2,
            "address": 33
          },
          {
            "tag": "getCPULoad",
            "type": "32int",
            "functionCode": 4,
            "objectsCount": 2,
            "address": 35
          }
        ]
      }
    ]
  },
  "id": "2c37dd3c-d968-4623-af70-b7c4b5b34a05",
  "logLevel": "INFO",
  "name": "modbus",
  "enableRemoteLogging": false
}

imbeacon, 2025-01-07:
As we can see - you have 50 configured datapoints, that should be polled each second, but it looks like it is not enough rate limits to send your data (You can read about rate limits for thingsboard.cloud - [here](https://thingsboard.io/docs/paas/subscription/#rate-limits)).

The gateway uses 80% of available rate limits, to have ability to get incoming RPC requests and shared attribute updates. 
So the reason of such behaviour is data that gateway stores in it’s storage. - When the gateway reads data from it - there are probably more than 1 message, with 50 datapoints, it tries to send it and then the client discovers that rate limit will be exceeded in case of sending and client waits before sending next batch of the data to ThingsBoard.

DuXiaoChuang, 2025-01-07:
> As we can see - you have 50 configured datapoints, that should be polled each second, but it looks like it is not enough rate limits to send your data (You can read about rate limits for [...]

what should i do to reslove this problem？？

imbeacon, 2025-01-07:
It looks like it is not enough rate limits for your gateway, what subscription type do you have on the cloud? 

I see 2 possible solutions:
1. Increase level of subscription.
2. Install locally instance of ThingsBoard, without any limitations.

Also, it may be a some kind of gateway issue, if your subscription have enough rate limits, but to be sure, we need to know what type of subscription you use.

DuXiaoChuang, 2025-01-07:
ok，thanks

---- Replied Message ----
| From | Illia ***@***.***> |
| Date | 01/07/2025 16:04 |
| To | ***@***.***> |
| Cc | Du ***@***.***>***@***.***> |
| Subject | Re: [thingsboard/thingsboard-gateway] [HELP]2024-12-25 10:55:40 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _wait_for_rate_limit_released - 749 - Rate limit reached for 1 seconds, waiting for rate limit to be released... (Issue #1635) |

It looks like it is not enough rate limits for your gateway, what subscription type do you have on the cloud?

I see 2 possible solutions:

Increase level of subscription.
Install locally instance of ThingsBoard, without any limitations.

Also, it may be a some kind of gateway issue, if your subscription have enough rate limits, but to be sure, we need to know what type of subscription you use.

closed 2025-01-10 by imbeacon