thingsboard/thingsboard-gateway #1640: [HELP]

franklupo, 2025-01-07:
**Describe the issue**
Through the gateway I read devices in MODBUS every 15 minutes and send to the cloud.
There were network problems and the communication stopped working.
The network started working again but it was not transmitting anymore. I had to restart the docker service and lost all the data.

**Error traceback (If it was raised):**
```
2025-01-06 22:40:21 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:21 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:32 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:32 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:32 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:43 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:43 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:43 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:43 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:53 - |INFO| - [slave.py] - slave - connect - 166 - Connected to IT50 - Pressa 25
2025-01-06 22:40:53 - |ERROR| - [modbus_connector.py] - modbus_connector - __poll_device - 239 - Socket is closed, connection is lost, for device IT50 - Pressa 25
2025-01-06 22:40:54 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:54 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:54 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:54 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:40:59 - |INFO| - [slave.py] - slave - connect - 166 - Connected to IT60 - Macchina 46
2025-01-06 22:40:59 - |ERROR| - [modbus_connector.py] - modbus_connector - __poll_device - 239 - Socket is closed, connection is lost, for device IT60 - Macchina 46
2025-01-06 22:41:05 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:16 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:21 - |INFO| - [slave.py] - slave - connect - 166 - Connected to IT60 - Uffici Piano Terra
2025-01-06 22:41:21 - |ERROR| - [modbus_connector.py] - modbus_connector - __poll_device - 239 - Socket is closed, connection is lost, for device IT60 - Uffici Piano Terra
2025-01-06 22:41:27 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:27 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:27 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:38 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:38 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:38 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:49 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:49 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:49 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:49 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:41:55 - |INFO| - [slave.py] - slave - connect - 166 - Connected to IT50 - Pressa 24
2025-01-06 22:41:55 - |ERROR| - [modbus_connector.py] - modbus_connector - __poll_device - 239 - Socket is closed, connection is lost, for device IT50 - Pressa 24
2025-01-06 22:42:00 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:11 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:11 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:22 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:22 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:22 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:33 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:44 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:55 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:55 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:55 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:55 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:42:57 - |WARNING| - [slave.py] - slave - connect - 160 - Maximum attempt count (5) for device "IT50 - Pressa 26" - encountered.
2025-01-06 22:42:57 - |ERROR| - [modbus_connector.py] - modbus_connector - __poll_device - 239 - Socket is closed, connection is lost, for device IT50 - Pressa 26
2025-01-06 22:43:06 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
[... cut ...]
2025-01-06 22:46:13 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:46:13 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
2025-01-06 22:46:13 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 925 - Queue size exceeded, waiting for messages to be processed by paho client.
```

**Versions:**
 - OS: Docker
 - Thingsboard IoT Gateway version 3.6.2

imbeacon, 2025-01-07:
Hi @franklupo,

Thank you for your interest in ThingsBoard IoT Gateway.

Please provide information about how many datapoints you try to read per time?

In general this issue relates to Paho mqtt client, that we are using as a base for our Python MQTT client. It means that mqtt client send messages to the platform, but they were not acknowledged and it didn’t remove them. We discovered behaviour like this in version 3.5, but it wasn’t appear in our tests after that version. So please provide, if possible, your configuration to give us ability to reproduce this situation.

franklupo, 2025-01-07:
do you mean modbus configuration?

Best regards

imbeacon, 2025-01-07:
Yes

franklupo, 2025-01-07:
[it60.json](https://github.com/user-attachments/files/18331701/it60.json)
[it50.json](https://github.com/user-attachments/files/18331702/it50.json)

Best reagrds

imbeacon, 2025-01-08:
Thank you for configuration, we will investigate this point, fix will be added in the one of next releases.
At the moment you can try to split devices to more connectors, like 2 devices per connector. But I’m not completely sure that it will help.

franklupo, 2025-01-27:
News?

imbeacon, 2025-03-21:
Hi @franklupo,

It is a complex issue, with the gateway and paho mqtt library, that is used in the gateway itself. From our side we tried to solve it in 3.7.2, but the part, related to the paho still not solved from their part, unfortunately. You can try to use 3.7.2 - it should work better.

closed 2025-03-21 by imbeacon