thingsboard/thingsboard-gateway #1641: [BUG] OPC UA Connector Fails to Automatically Reconnect After OPC UA Source Server Restart

sanjayN4497, 2025-01-07:
**Describe**
We are experiencing a critical issue with the OPC UA connector in ThingsBoard Gateway 3.6.4. After the OPC UA Source server is restarted, the OPC UA integration fails to automatically reconnect to the OPC UA server. The only way to restore the connection is by manually re-deploy/re-configure the OPC UA integration.

This behavior is problematic in scenarios where a seamless, uninterrupted connection to the OPC UA server is crucial for continuous data collection. Ideally, the OPC UA connector should automatically attempt to reconnect to the server after detecting a disconnection.

**Connector name (If bug in the some connector):**
OPC-UA Connector

**Error traceback (If available):**

Observed Behavior:
1) After restarting the OPC UA source server, the OPC UA connector does not attempt to reconnect automatically.
2) If the ThingsBoard Gateway is running in a Docker container and the container is restarted twice manually, the connection may eventually be re-established.
3) This behavior introduces delays and requires manual intervention to restore functionality with either docker restart or reconfigure connection.

Expected Behavior:
1) The OPC UA connector should automatically detect when the OPC UA server is unavailable and continuously attempt to reconnect.
2) No manual intervention should be required to reconfigure or restart the integration.
3) The connector should pull data continuously, regardless of whether the OPC UA server restarts or temporarily goes offline.

**Versions (please complete the following information):**
 - OS: Ubuntu
 - Thingsboard IoT Gateway version  3.6.4
[image]

ankushdineshrana, 2025-01-13:
I am also facing same issue, please help @imbeacon  @samson0v

imbeacon, 2025-03-21:
Hi,

Please try to use the latest version of the gateway (3.7.2). The gateway had several issues with reconnection, but they should be solved in latest releases. Feel free to open a new issues if you discover some misbehaving.

closed 2025-03-21 by imbeacon