thingsboard/thingsboard-gateway #1649: [HELP]

evermore-4, 2025-01-13:
**Describe the issue**
I have a modbus serial configured, yet the gateway fails to configure a slave device complaining about ` [tb_logger.py] - tb_logger exception - 162 - Failed to add slave: 'host' `.

**Configuration**
[config.json](https://github.com/user-attachments/files/18399596/config.json)
```json
{
  "name": "Name",
  "enableRemoteLogging": true,
  "id": "68aa1250-7b3e-4ee6-9f4e-094b2dfa6dd2",
  "master": {
    "slaves": [
      {
        "method": "rtu",
        "baudrate": 9600,
        "stopbits": 1,
        "bytesize": 8,
        "parity": "N",
        "strict": false,
        "unitId": 31,
        "deviceName": "Device Name",
        "deviceType": "Device Type",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "reportStrategy": {
          "type": "ON_CHANGE_OR_REPORT_PERIOD",
          "reportPeriod": 30000
        },
        "type": "serial",
        "attributes": [
          {
            "tag": "Speed Refernce",
            "type": "bytes",
            "address": 640,
            "objectsCount": 16,
            "functionCode": 3
          }
        ],
        "timeseries": [
          {
            "tag": "Run Data",
            "type": "bytes",
            "address": 64,
            "objectsCount": 16,
            "functionCode": 3
          }
        ],
        "attributeUpdates": [],
        "rpc": [],
        "port": "/dev/ttyS0"
      }
    ]
  }
}

```
[docker-compose.txt](https://github.com/user-attachments/files/18399753/docker-compose.txt)
```yaml
version: '3.6.3'
services:
  # ThingsBoard IoT Gateway Service Configuration
  tb-gateway:
    image: thingsboard/tb-gateway
    container_name: tb-gateway
    restart: always

    # Ports bindings - required by some connectors
    ports:
      #- "5000:5000" # Comment if you don't use REST connector and change if you use another port
        # Uncomment and modify the following ports based on connector usage:
#        - "1052:1052" # BACnet connector
        - "5026:5026" # Modbus TCP connector (Modbus Slave)
#        - "50000:50000/tcp" # Socket connector with type TCP
#        - "50000:50000/udp" # Socket connector with type UDP

    # Necessary mapping for Linux
    extra_hosts:
      - "host.docker.internal:host-gateway"

    # Environment variables
    environment:
      - host=host.docker.internal
      - port=1883
      - accessToken=REDACTED

    # Volumes bind
    volumes:
      - tb-gw-config:/thingsboard_gateway/config
      - tb-gw-logs:/thingsboard_gateway/logs
      - tb-gw-extensions:/thingsboard_gateway/extensions
    devices:
      - /dev/ttyS0
      - /dev/ttyS1
      - /dev/ttyS2
      - /dev/ttyS3
    
# Volumes declaration for configurations, extensions and configuration
volumes:
  tb-gw-config:
    name: tb-gw-config
  tb-gw-logs:
    name: tb-gw-logs
  tb-gw-extensions:
    name: tb-gw-extensions
```

**Connector name (If you need help with some connector/converter):**
Modbus Connector

**Error traceback (If it was raised):**
Took me a while to find this in tb_connection.log
```
Traceback (most recent call last):
  File \"/thingsboard_gateway/connectors/modbus/modbus_connector.py\", line 204, in __add_slaves
    self.__add_slave(slave_config)
  File \"/thingsboard_gateway/connectors/modbus/modbus_connector.py\", line 195, in __add_slave
    slave = Slave(self, self.__log, slave_config)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File \"/thingsboard_gateway/connectors/modbus/slave.py\", line 39, in __init__
    config[\'host\']) + \":\" + str(config[\'port\']) + \' \' + config[\'deviceName\']
    ~~~~~~^^^^^^^^
KeyError: \'host\'
```

**Versions (please complete the following information):**
 - Host OS: Ubuntu 24.10
 - Python version 3.12.7

imbeacon, 2025-01-14:
Hi @evermore-4,

Thank you for your interest in ThingsBoard IoT Gateway.
The issue, that you report, was in 3.6.2, but was fixed in 3.6.3.
It looks like you have older image, could you try to set the following parameter for image and try again:
`image: thingsboard/tb-gateway:3.6.3`

evermore-4, 2025-01-14:
That worked, partially.
Now I'm getting this:
 [modbus_connector.py] - modbus_connector __poll_device - 249 - Socket is closed, connection is lost, for device Device Name

evermore-4, 2025-01-14:
[ConnectorA_connector.log](https://github.com/user-attachments/files/18416271/ConnectorA_connector.log)
[extension.log](https://github.com/user-attachments/files/18416272/extension.log)
[service.log](https://github.com/user-attachments/files/18416273/service.log)
[storage.log](https://github.com/user-attachments/files/18416274/storage.log)
[tb_connection.log](https://github.com/user-attachments/files/18416275/tb_connection.log)

aelde, 2025-01-27:
same problem. did it solve yet

imbeacon, 2025-01-29:
Hi,

Several issues were identified related to locks, port management, and data reading. Unfortunately, I’m unable to update the Docker image at this time, as new features have already been added to the repository.

The fix is available in the master branch and will be included in the next release, which we expect to publish around February 6.

closed 2025-02-06 by imbeacon