thingsboard/thingsboard-gateway #1652: [HELP] Failed to use REST connector in window os

No-Ten, 2025-01-15:
Hi，I use the release 3.6.3 source code to build the gateway in window 10 OS. Edge is running and has been connected to the gateway. Other protocols such as MQTT, Modbus, and OPCUA are working normally, except for the REST protocol which has failed.

However, when I run the same code in a Linux environment, the REST protocol connector can run normally.

So I don't know if it's a problem with my window os environment or something else?

Edge connectos page
[image]

idea console log

[image]

tb_gateway.json
```json
{
  "thingsboard": {
    "host": "localhost",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "latencyDebugMode": false,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 8196,
    "minPackSendDelayMS": 50,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "QY8eAFdisoV43fqQS50p"
    },
    "qos": 1,
    "reportStrategy": {
      "type": "ON_RECEIVED"
    },
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    },
    "rateLimits": "DEFAULT_TELEMETRY_RATE_LIMIT",
    "dpRateLimits": "DEFAULT_TELEMETRY_DP_RATE_LIMIT",
    "messagesRateLimits": "DEFAULT_MESSAGES_RATE_LIMIT",
    "deviceMessagesRateLimits": "DEFAULT_MESSAGES_RATE_LIMIT",
    "deviceRateLimits": "DEFAULT_TELEMETRY_RATE_LIMIT",
    "deviceDpRateLimits": "DEFAULT_TELEMETRY_DP_RATE_LIMIT",
    "ts": 1736838929172
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7,
    "ts": 1736838929172
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepAliveTimeMs": 10001,
    "keepAliveTimeoutMs": 5000,
    "keepAlivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000
  },
  "connectors": [
    {
      "type": "rest",
      "name": "REST",
      "configuration": "rest.json"
    }
  ]
}
```

rest.json
```json

{
  "host": "127.0.0.1",
  "port": 5000,
  "SSL": false,
  "mapping": [
    {
      "endpoint": "/test_device",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "basic",
        "username": "username",
        "password": "passwword"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "Rest-device",
        "deviceTypeExpression": "default",
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    }
  ],
  "logLevel": "DEBUG",
  "name": "REST",
  "enableRemoteLogging": true,
  "id": "99109c88-b923-45e6-afbe-44ea4ef5cfff",
  "configVersion": "3.6.3"
}

```

error log
```bash

2025-01-15 13:58:03 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 246 - Configuration update request received.
2025-01-15 13:58:03 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 246 - Configuration update request received.
2025-01-15 13:58:04 - |INFO| - [tb_loader.py] - tb_loader - import_module - 73 - Import RESTConnector from E:\temp\tb-code\thingsboard-gateway-3.6.3\thingsboard_gateway\connectors\rest.
2025-01-15 13:58:04 - |INFO| - [tb_loader.py] - tb_loader - import_module - 73 - Import JsonRESTDownlinkConverter from E:\temp\tb-code\thingsboard-gateway-3.6.3\thingsboard_gateway\connectors\rest.
2025-01-15 13:58:04 - |INFO| - [tb_loader.py] - tb_loader - import_module - 73 - Import JsonRESTUplinkConverter from E:\temp\tb-code\thingsboard-gateway-3.6.3\thingsboard_gateway\connectors\rest.
2025-01-15 13:58:04 - |DEBUG| - [rest_connector.py] - rest_connector - __fill_requests_from_TB - 366 - Requests from TB: {'attributeUpdates': [], 'serverSideRpc': []}
2025-01-15 13:58:04 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __connect_with_connectors - 1024 - Cannot find separated report strategy for connector 'REST'.                                                          The main report strategy                                                          will be used as a connector report strategy.
2025-01-15 13:58:04 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 246 - Configuration update request received.
Traceback (most recent call last):
  File "E:\temp\tb-code\thingsboard-gateway-3.6.3\thingsboard_gateway\connectors\rest\rest_connector.py", line 185, in __run_server
    await site.start()
  File "E:\temp\tb-code\thingsboard-gateway-3.6.3\venv\lib\site-packages\aiohttp\web_runner.py", line 121, in start
    self._server = await loop.create_server(
  File "E:\Python\Python39\lib\asyncio\base_events.py", line 1493, in create_server
    _set_reuseport(sock)
  File "E:\Python\Python39\lib\asyncio\base_events.py", line 94, in _set_reuseport
    raise ValueError('reuse_port not supported by socket module')
ValueError: reuse_port not supported by socket module
Exception in thread REST:
Traceback (most recent call last):
  File "E:\Python\Python39\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "E:\temp\tb-code\thingsboard-gateway-3.6.3\thingsboard_gateway\connectors\rest\rest_connector.py", line 194, in run
    self._loop.run_until_complete(self.__run_server())
  File "E:\Python\Python39\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "E:\temp\tb-code\thingsboard-gateway-3.6.3\thingsboard_gateway\connectors\rest\rest_connector.py", line 185, in __run_server
    await site.start()
  File "E:\temp\tb-code\thingsboard-gateway-3.6.3\venv\lib\site-packages\aiohttp\web_runner.py", line 121, in start
    self._server = await loop.create_server(
  File "E:\Python\Python39\lib\asyncio\base_events.py", line 1493, in create_server
    _set_reuseport(sock)
  File "E:\Python\Python39\lib\asyncio\base_events.py", line 94, in _set_reuseport
    raise ValueError('reuse_port not supported by socket module')
ValueError: reuse_port not supported by socket module

```

**Versions**
 - OS: win10
 - Thingsboard IoT Gateway version 3.6.3
 - Python version 3.9.11
 - Edge: 3.8

imbeacon, 2025-01-15:
Hi @No-Ten,

Thank you for your interest in ThingsBoard IoT Gateway, we have added a fix for this situation, please try to use a version from the master branch and let us know about the results.

No-Ten, 2025-01-15:
> Hi @No-Ten,
> 
> Thank you for your interest in ThingsBoard IoT Gateway, we have added a fix for this situation, please try to use a version from the master branch and let us know about the results.

So cool! It has been resolved. Thank you !!!

closed 2025-01-15 by imbeacon