thingsboard/thingsboard-gateway #1654: [HELP] Connectors not working / enabled

go4energy, 2025-01-15:
**Describe the issue**
Hello, I have issues setting up the Modbus TCP connector in the Gateway.

First there are some general questions, which I could not answer reading the rocs.

I setup an TB edge (docker compose), and on the same device the IoT gateway (also docker compose).
The TB server CE is running on another device.
That basically works.

What is the correct way to setup the IoT gateway to connect to the edge? I did not find a way in the Edge UI to add/setup a gateway.
What I did:
I went to the TB cloud demo and exported the dashboard "ThingsBoard IoT Gateways".
I imported this to the TB server CE, and than assigned this Dashboard to the Edge.
At the Edge, using that Dashboard I added an IoT Gateway. I got the docker-compose.yml from there and successfully started it (after some adjustment in the both docker-compose files for Edge and IoT gateway).

**Is that the right way to do it?**

Gateway shows as "Active" in the Edge Dashbaord, "Enabled Connectors = 0".
On the Edge IoT Gateways Dashboard, I can't setup the connectors. It gives me:

_```
Widget wurde aufgrund der folgenden Fehler nicht geladen:
Unable to load module from url: tb-resource;/api/resource/js_module/system/gateway-management-extension.js
```_

I can edit the connectors on the TB server, where I see "Enabled Connectors = 1":

<img width="1243" alt="image" src="https://github.com/user-attachments/assets/88f0c3ef-2244-463b-a714-a417f4cdfb79" />

Why is it always out of sync?
Should the configuration be automatically be updated on the IoT gateway once I save the config?
Do I need to restart the docker-compose for the IoT gateway?

Also I noticed that I can't add my timeseries from the IoT gateway to a dashboard. That should be ok for now as there is no connector running?

So for now, I don't even know if I ask the correct questions.
Still need some clarification about the whole interaction of server, cloud and IoT gateway.

Thanks for your support!

**Connector name (If you need help with some connector/converter):**
MODBUS TCP (master)

**Versions (please complete the following information):**
 - OS: Ubuntu 24.04 for TB Server CE
          DietPi v8.22.3 for TB Edge CE
 - Thingsboard IoT Gateway version e.g. 3.6.3 (running on Edge)
 - Python version[e.g. 3.9]

<img width="1178" alt="image" src="https://github.com/user-attachments/assets/424302b6-8003-4b84-8a2c-ae5c114e4b6b" />

imbeacon, 2025-02-05:
Hi all,

Thank you for your interest in ThingsBoard IoT Gateway. It looks like this issue relates more to the Edge implementation. @volodymyr-babak please help with this issue.

tGallimberti-Kerberos, 2025-02-05:
Hi @imbeacon just talked with Edge support and in my case was a problem of the evaluation license limited to 5 devices

closed 2025-03-21 by imbeacon