thingsboard/thingsboard-gateway #1658: BACnet Connector

jmorbitzerpke, 2025-01-20:
Good morning!

I have problems connecting to a BACnet device via TB-Gateway. Which configuration is the right one? There is a difference in describing the datapoint.

Configuration shown in Git:

"devices": [
    {
      "deviceInfo": {
        "deviceNameExpressionSource": "expression",
        "deviceNameExpression": "BACnet Device ${objectName}",
        "deviceProfileExpressionSource": "constant",
        "deviceProfileExpression": "default"
      },
      "address": "192.168.0.102:47808",
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "temperature",
          "objectType": "analogInput",
          "objectId": "0",
          "propertyId": "presentValue"
        }
      ],
      "timeseries": [
        {
          "key": "water_temp",
          "objectType": "analogInput",
          "objectId": "1",
          "propertyId": "presentValue"
        },

Configuration in TB-Gateway (auto generated):

"devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.0.102:47808",
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "temperature",
          "type": "string",
          "objectId": "analogOutput:1",
          "propertyId": "presentValue"
        }
      ],
      "timeseries": [
        {
          "key": "state",
          "type": "bool",
          "objectId": "binaryValue:1",
          "propertyId": "presentValue"
        }

samson0v, 2025-01-20:
Hi @jmorbitzerpke, thanks for your interest in ThingsBoard IoT Gateway!
The first one is in the right format.
Also, if you have a problem with BACnet connectivity, send us logs with the DEBUG level.

jmorbitzerpke, 2025-01-20:
Here are some screenshots. I got a device at 192.168.0.102 with AV1. Connector doesn't seem to start, because under BACnet connectors there is no entry. 

[image]

[image]

[image]

[image]

jmorbitzerpke, 2025-01-20:
Could you please supply a working config file? What do you need to know about the BACnet device.

Some information : 

1.  IP-Gateway: 192.168.0.20
2. IP-BACnet device: 192.168.0.102
3. Port: 47808
4. Data: Analog Value, AV1
5. Installation on a RPi CM4 with version 3.6.1

Thanks

jmorbitzerpke, 2025-01-23:
Hello!

No one facing the same problem?

jmorbitzerpke, 2025-01-27:
[image]

imbeacon, 2025-01-29:
Hi @jmorbitzerpke,

This situation is caused, because we cannot discover which docker interface will be used to retrieve response from device.
In order to fix this issue - you can use special parameter - altResponsesAddresses with value from log “Received APDU, from ***. trying to find device…". Your configuration will look like:

```
"devices": [
{
"deviceInfo": {
"deviceNameExpressionSource": "expression",
"deviceNameExpression": "BACnet Device ${objectName}",
"deviceProfileExpressionSource": "constant",
"deviceProfileExpression": "default"
},
"address": "192.168.0.102:47808”,
"altResponsesAddresses": ["172.18.0.1:51018"],
"pollPeriod": 10000,
"attributes": [
{
"key": "temperature",
"objectType": "analogInput",
"objectId": "0",
"propertyId": "presentValue"
}
],
"timeseries": [
{
"key": "water_temp",
"objectType": "analogInput",
"objectId": "1",
"propertyId": "presentValue"
},
```

jmorbitzerpke, 2025-01-29:
Thank you for your answer. But there is one problem. The port seems to be randomly chosen by the connector when enabled. So when I change the config to the port dispayed, the port changes after the config has changed. Is there a way to bind specific port?

imbeacon, 2025-02-06:
Hi @jmorbitzerpke,

You are right at his point, you can try to set network to host (`network_mode:host` in docker-compose.yaml) and remove altResponsesAddresses. Unfortunately it looks for me now, like only one possible way to correctly handle requests from different devices. because in other cases, when the docker uses bridged network - all requests, even from different devices will arrive from the same ip address.

pke-plorenz, 2025-02-14:
hi @imbeacon 
while troubleshooting this with @jmorbitzerpke, I discovered we were using tb-gateway version 3.6.1.
Since the latest three releases contain bugfixes regarding bacnet we updated to version 3.7.0 and it seems like the issue is resolved.

However, the latest rpi-compatible image on docker hub is version 3.6.1, so we built the image ourselves.
Is there a reason rpi images are no longer published / would it be possible to publish rpi images again?

imbeacon, 2025-07-08:
Hi,

The new version is released, also, images for RPi also available on DockerHub.

Please try to use it, feel free to open new issue if you discover something.

closed 2025-07-08 by imbeacon