thingsboard/thingsboard-gateway #1659: [HELP] |WARNING| - [odbc_connector.py] - odbc_connector - __process_row - 289 - [YM_ODBC] Failed to process database row: 'ConvertedData' object has no attribute 'items'

kpdqzxl1975, 2025-01-21:
F:\thingsboard-gateway-3.6.3\.venv\Scripts\python.exe F:\thingsboard-gateway-3.6.3\thingsboard_gateway\tb_gateway.py 
2025-01-21 15:41:30 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 189 - Gateway starting...
2025-01-21 15:41:32 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 200 - ThingsBoard IoT gateway version: 3.6.3
2025-01-21 15:41:32 - |INFO| - [tb_loader.py] - tb_loader - import_module - 73 - Import OdbcConnector from F:\thingsboard-gateway-3.6.3\thingsboard_gateway\connectors\odbc.
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __configure_pyodbc - 437 - [YM_ODBC] Set pyodbc attributes: {'pooling': False}
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __parse_rpc_config - 445 - [YM_ODBC] Processing unknown RPC enabled
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __parse_rpc_config - 451 - [YM_ODBC] Overriding RPC config disabled
2025-01-21 15:41:32 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __connect_with_connectors - 1024 - Cannot find separated report strategy for connector 'YM_ODBC'.                                                          The main report strategy                                                          will be used as a connector report strategy.
2025-01-21 15:41:32 - |DEBUG| - [odbc_connector.py] - odbc_connector - open - 96 - [YM_ODBC] Starting...
2025-01-21 15:41:32 - |DEBUG| - [odbc_connector.py] - odbc_connector - __init_connection - 308 - [YM_ODBC] Opening connection to database
2025-01-21 15:41:32 - |INFO| - [tb_client.py] - tb_client - connect - 375 - Connecting to ThingsBoard...
2025-01-21 15:41:32 - |INFO| - [tb_client.py] - tb_client - __send_connect - 402 - Sending connect to 127.0.0.1:1883, tls: False...
2025-01-21 15:41:32 - |WARNING| - [tb_client.py] - tb_client - connect - 386 - Cannot use TLS connection on this port. Client will try to connect without TLS.
2025-01-21 15:41:32 - |INFO| - [tb_client.py] - tb_client - __send_connect - 402 - Sending connect to 127.0.0.1:1883, tls: False...
2025-01-21 15:41:32 - |INFO| - [tb_client.py] - tb_client - __send_connect - 405 - Connect msg sent to platform.
2025-01-21 15:41:32 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 436 - MQTT client <paho.mqtt.client.Client object at 0x00000155226873B0> - Connected!
2025-01-21 15:41:32 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 738 - Service configuration was successfully retrieved and applied.
2025-01-21 15:41:32 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 739 - Current limits: {'rateLimit': {'messages': None, 'telemetryMessages': None, 'telemetryDataPoints': None}, 'maxPayloadSize': 65536, 'maxInflightMessages': 100}
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __init_connection - 312 - [YM_ODBC] Setting encoding to utf-8
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __init_connection - 319 - [YM_ODBC] Setting SQL_CHAR decoding to utf-8
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __init_connection - 322 - [YM_ODBC] Setting SQL_WCHAR decoding to utf-8
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __init_connection - 326 - [YM_ODBC] Setting SQL_WMETADATA decoding to utf-16le
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __init_connection - 335 - [YM_ODBC] Connection to database opened, attributes {'autocommit': True, 'timeout': 0}
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __init_iterator - 368 - [YM_ODBC] Iterator saving disabled
2025-01-21 15:41:32 - |INFO| - [odbc_connector.py] - odbc_connector - __init_iterator - 387 - [YM_ODBC] Init iterator from database: column=ts, start_value=1735689599999
2025-01-21 15:41:33 - |INFO| - [odbc_connector.py] - odbc_connector - __poll - 245 - [YM_ODBC] Fetch column names: ['bool_v', 'str_v', 'dbl_v', 'long_v', 'entity_id', 'ts']
2025-01-21 15:41:33 - |DEBUG| - [odbc_connector.py] - odbc_connector - __poll - 250 - [YM_ODBC] Fetch row: (None, None, None, 66686, '1B9A87F0-D191-11EF-93BF-C932F7624471', 1735689600000)
2025-01-21 15:41:33 - |ERROR| - [odbc_uplink_converter.py] - odbc_uplink_converter - convert - 114 - Failed to convert SQL data to TB format: Invalid arguments for add_attribute
2025-01-21 15:41:33 - |WARNING| - [odbc_connector.py] - odbc_connector - __process_row - 289 - [YM_ODBC] Failed to process database row: 'ConvertedData' object has no attribute 'items'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

kpdqzxl1975, 2025-01-21:
converted_data = self.__converter.convert(self.__config["mapping"], data)
for device_name, data in converted_data.items():
AttributeError("'ConvertedData' object has no attribute 'items'")

imbeacon, 2025-01-21:
Hi @kpdqzxl1975,

Please add converter configuration section from your odbc connector config file.

imbeacon, 2025-01-23:
Hi @kpdqzxl1975,

Please try the version from the master branch, there is a fix available there.

kpdqzxl1975, 2025-01-24:
Modified the code.:    def __process_row(self, row):
        try:
            data = self.row_to_dict(row)

            converted_data = self.__converter.convert(self.__config["mapping"], data)

            # for device_name, data in converted_data.items():
            #
            #     StatisticsService.count_connector_message(self._log.name, 'convertersAttrProduced',
            #                                               count=data.attributes_datapoints_count)
            #     StatisticsService.count_connector_message(self._log.name, 'convertersTsProduced',
            #                                               count=data.telemetry_datapoints_count)
            #
            #     data.device_name = eval(self.__config["mapping"]["device"]["name"], globals(), data)
            #
            #     device_type = eval(self.__config["mapping"]["device"]["type"], globals(), data)
            #     if not device_type:
            #         device_type = self.__config["mapping"]["device"].get("type", "default")
            #     data.device_type = device_type
            #
            #     if data.telemetry_datapoints_count + data.attributes_datapoints_count > 0:
            #         self.__iterator["value"] = getattr(row, self.__iterator["name"])
            #         self.__check_and_send(data)
            StatisticsService.count_connector_message(self._log.name, 'convertersAttrProduced',
                                                      count=converted_data.attributes_datapoints_count)
            StatisticsService.count_connector_message(self._log.name, 'convertersTsProduced',
                                                      count=converted_data.telemetry_datapoints_count)

            # 直接处理 converted_data 对象中的 device_name 和 device_type 属性
            converted_data.device_name = eval(self.__config["mapping"]["device"]["name"], globals(), data)
            # device_type = eval(self.__config["mapping"]["device"]["type"], globals(), data)
            # if not device_type:
            device_type = self.__config["mapping"]["device"].get("type", "default")
            converted_data.device_type = device_type

            if converted_data.telemetry_datapoints_count + converted_data.attributes_datapoints_count > 0:
                self.__iterator["value"] = getattr(row, self.__iterator["name"])
                self.__check_and_send(converted_data)  # 直接发送 converted_data 实例
        except Exception as e:
            self._log.warning("[%s] Failed to process database row: %s", self.get_name(), str(e))

kpdqzxl1975, 2025-01-24:
ok!

closed 2025-02-06 by imbeacon