thingsboard/thingsboard-gateway #1673: [BUG] 3.7 KeyError: 'objectsCount'

pro100805, 2025-02-08:
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; preset: enabled)
     Active: active (running) since Sat 2025-02-08 15:55:13 CET; 7min ago
   Main PID: 904 (python3)
      Tasks: 42 (limit: 2211)
     Memory: 59.3M (peak: 71.0M)
        CPU: 8min 1.218s
     CGroup: /system.slice/thingsboard-gateway.service
             └─904 /var/lib/thingsboard_gateway/venv/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

Feb 08 16:02:13 thingsboard-server python3[904]:   File "/usr/lib/python3.12/asyncio/events.py", line 88, in _run
Feb 08 16:02:13 thingsboard-server python3[904]:     self._context.run(self._callback, *self._args)
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 229, in __process_requests
Feb 08 16:02:13 thingsboard-server python3[904]:     await self.__poll_device(slave)
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 260, in __poll_device
Feb 08 16:02:13 thingsboard-server python3[904]:     self.__log.error('Failed to poll %s device: %s', slave, exc_info=e)
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/tb_utility/tb_logger.py", line 172, in error
Feb 08 16:02:13 thingsboard-server python3[904]:     super(TbLogger, self).error(msg, *args, **kwargs)
Feb 08 16:02:13 thingsboard-server python3[904]: Message: 'Failed to poll %s device: %s'
Feb 08 16:02:13 thingsboard-server python3[904]: Arguments: (<Slave(Modbus slave processor for unit 1 on host 192.168.18.11:502 Analog Input, started daemon 269866573558144)>,)
Feb 08 16:02:13 thingsboard-server python3[904]: --- Logging error ---
Feb 08 16:02:13 thingsboard-server python3[904]: Traceback (most recent call last):
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 247, in __poll_device
Feb 08 16:02:13 thingsboard-server python3[904]:     slave_data = await self.__read_slave_data(slave)
Feb 08 16:02:13 thingsboard-server python3[904]:                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 273, in __read_slave_data
Feb 08 16:02:13 thingsboard-server python3[904]:     response = await slave.read(config['functionCode'], config['address'], config['objectsCount'])
Feb 08 16:02:13 thingsboard-server python3[904]:                                                                            ~~~~~~^^^^^^^^^^^^^^^^
Feb 08 16:02:13 thingsboard-server python3[904]: KeyError: 'objectsCount'
Feb 08 16:02:13 thingsboard-server python3[904]: During handling of the above exception, another exception occurred:
Feb 08 16:02:13 thingsboard-server python3[904]: Traceback (most recent call last):
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/usr/lib/python3.12/logging/__init__.py", line 1160, in emit
Feb 08 16:02:13 thingsboard-server python3[904]:     msg = self.format(record)
Feb 08 16:02:13 thingsboard-server python3[904]:           ^^^^^^^^^^^^^^^^^^^
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/usr/lib/python3.12/logging/__init__.py", line 999, in format
Feb 08 16:02:13 thingsboard-server python3[904]:     return fmt.format(record)
Feb 08 16:02:13 thingsboard-server python3[904]:            ^^^^^^^^^^^^^^^^^^
Feb 08 16:02:13 thingsboard-server python3[904]:   File "/usr/lib/python3.12/logging/__init__.py", line 703, in format
Feb 08 16:02:13 thingsboard-server python3[904]:     record.message = record.getMessage()
Feb 08 16:02:13 thingsboard-server python3[904]:                      ^^^^^^^^^^^^^^^^^^^

Clear install 

 - OS: Ubuntu Server 24.01
 - Thingsboard IoT 3.9
 - Gateway version 3.7
 - Python version 3.12

imbeacon, 2025-02-10:
Hi @pro100805,

It looks like “objectsCount” parameter is missing in your configuration, could you provide your configuration for investigation?

pro100805, 2025-02-11:
Hello, thank you for your quick reply.
I have a separate TB 3.9 server that works fine with GW 3.5.1
I'm doing a clean install since I can't upgrade, and I'm using all the same configuration files that worked with 3.5.1 making changes only in tb_gateway.json (token, server IP) connector configuration on the TB side.
devices are visible and active but there is no data transfer

[serialUsb.json.txt](https://github.com/user-attachments/files/18749826/serialUsb.json.txt)
[tb_gateway.json.txt](https://github.com/user-attachments/files/18749827/tb_gateway.json.txt)

pro100805, 2025-02-11:
In 3.5.1 RPC function Get and Set with this configuration work normally sometimes with not big delays

pro100805, 2025-03-07:
[image]

imbeacon, 2025-03-21:
Hi @pro100805,

Unfortunately, it is a specific of the widget, we have passed this information to ThingsBoard product team. They will think on how to solve this issue. it appears because this widget is only one, which cannot handle different types of data, except primitives. At the moment, it is not possible to use Blob widget with the gateway, unfortunately. But you can avoid using RPC in this case and use shared attribute update functionality. It also give you ability to synchronise states between the device and ThingsBoard. Because when device connects - the gateway requests it's shared attributes and passes to device, to synchronise states of registers and values in them (At the moment implemented only for Modbus connector).

According to the main issue - I think it will be a complete redesign of the widget and it's configuration, then it will be able to handle values different from primitives.

closed 2025-03-21 by imbeacon