thingsboard/thingsboard-gateway #1677: [HELP] Unable to connect to a OPCUA server with Basic256Sha256

andersenthomas98, 2025-02-13:
I am not able to connect to a OPCUA server with Basic256Sha256 security policy due to the error `BadCertificateInvalidUri(2148990976)` 

According to [this](https://github.com/thingsboard/thingsboard-gateway/issues/746) old issue I might be missing an application uri, although I cannot find it documented anywhere. However, adding the uri to the OPCUA connector configuration does not seem to help either.

Here is my OPCUA connector configuration
```
{
  "server": {
    "url": "opc.tcp://*****",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 3600000,
    "enableSubscriptions": false,
    "subCheckPeriodInMillis": 60000,
    "showMap": true,
    "security": "Basic256Sha256",
    "identity": {
      "type": "cert.PEM",
      "cert": "/thingsboard_gateway/config/credentials/kepserver/tbgw_cert.pem",
      "privateKey": "/thingsboard_gateway/config/credentials/kepserver/tbgw_key.pem",
      "mode": "SignAndEncrypt",
      "username": "******",
      "password": "******"
    },
    "pollPeriodInMillis": 60000
  },
  ...
}

```

Error:

[image]

**Versions:**
 - Docker: 27.3.1 
 - Thingsboard IoT Gateway version: 3.6.3

andersenthomas98, 2025-02-18:
FYI, I am trying to connect to KepserverEx.

samson0v, 2025-05-13:
Hi @andersenthomas98, thanks for your interest in ThingsBoard IoT Gateway!
We are trying to reproduce your issue with Prosys OPC-UA Simulation Server, and everything works fine.
As you can see from the error, you should provide the Certificate URI.
Also, set CA Certificate in the [connector config section](https://thingsboard.io/docs/iot-gateway/config/opc-ua/?opcuasecuritysubsection=advanced&opcuaIdentityConfig=certpem#subsection-security).

[image]

kpdqzxl, 2025-07-16:
ca.pem怎么生成，怎么连接kepserverex

kpdqzxl, 2025-07-16:
证书 URI怎么提供，我是连接KepserverEx，仍然没有解决；[opcua_connector.py] - opcua_connector retry_connect_with_backoff - 443 - Encountered error: BadSecurityChecksFailed(2148728832). Next connection try in 128 second(s)...

kpdqzxl, 2025-07-23:
application uri在哪设置？