thingsboard/thingsboard-gateway #1678: [BUG] High CPU usage without any active connector

nightcore500, 2025-02-14:
I am observing an issue with a freshly set up ThingsBoard Gateway (3.7.0) running in an LXC container with Debian 12.9 & Python 3.11.2.
The gateway is currently only connected to ThingsBoard. No connectors are active at the moment.
Nevertheless, the Python process is causing 100% CPU load:
```
    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                    
   4995 thingsb+  20   0 1129920  57600  13440 S 100.7   2.7   6:08.44 python3
```

   
```
Feb 14 10:56:39 srv-tbiotgateway systemd[1]: Started thingsboard-gateway.service - ThingsBoard Gateway.
Feb 14 10:56:39 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:39 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 188 - Gateway starting...
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 199 - ThingsBoard IoT gateway version: 3.7.0
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 976 - Connectors - not found, waiting for remote configuration.
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_client.py] - tb_client - connect - 376 - Connecting to ThingsBoard...
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_client.py] - tb_client - __send_connect - 403 - Sending connect to 10.30.2.211:1883, tls: False...
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |WARNING| - [tb_client.py] - tb_client - connect - 387 - Cannot use TLS connection on this port. Client will try to connect without TLS.
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_client.py] - tb_client - __send_connect - 403 - Sending connect to 10.30.2.211:1883, tls: False...
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_client.py] - tb_client - __send_connect - 406 - Connect msg sent to platform.
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 436 - MQTT client <paho.mqtt.client.Client object at 0x7f0a2bfa0fd0> - Connected!
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 738 - Service configuration was successfully retrieved and applied.
Feb 14 10:56:40 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:40 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 739 - Current limits: {'rateLimits': {'messages': None, 'telemetryMessages': None, 'telemetryDataPoints': None}, 'maxPayloadSize': 65536, 'maxInflightMessages': 100}
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 270 - Gateway core started.
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 255 - Subscribed to *|* with id 1 for device *
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - __init__ - 83 - Remote Configurator started
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 976 - Connectors - not found, waiting for remote configuration.
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __process_remote_logging_update - 702 - Remote logging level set to: INFO
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 287 - Gateway connectors initialized.
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 291 - Persistent devices loaded.
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 246 - Configuration update request received.
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 315 - Gateway started.
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |ERROR| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_logs_configuration_update - 447 - Cannot create handler databaseHandler with [Errno 2] No such file or directory: './logs/database.log' error. Skipping...
Feb 14 10:56:41 srv-tbiotgateway python3[5111]: 2025-02-14 10:56:41 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 207 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 81.98, 'diskUsage': 6.3, 'freeMemory': 3.8, 'freeDisk': '7.5G', 'gwProcessCpuUsage': 25.4, 'gwMemory': 2.7, 'gwProcessMemoryFullInfo': {'rss': 58589184, 'vms': 1157046272, 'shared': 13762560, 'text': 2838528, 'lib': 0, 'data': 174149632, 'dirty': 0, 'uss': 54829056, 'pss': 55899136, 'swap': 3.8, 'freeDisk': '7.5G', 'gwProcessCpuUsage': 25.4, 'gwMemory': 2.7, 'gwProcessMemoryFullInfo': {'rss': 58589184, 'vms': 1157046272, 'shared': 13762560, 'text': 2838528, 'lib': 0, 'data': 174149632, 'dirty': 0, 'uss': 54829056, 'pss': 55899136, 'swap': 0}, 'msgsSentToPlatform': 12, 'msgsReceivedFromPlatform': 1}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 0, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {}}

```

igorjan, 2025-02-27:
Did You manage to find a solution for this I seem to have the exact same problem?

GeXin666, 2025-03-07:
docker run \
--name tb-gateway \
-e host=192.168.80.101 \
-e port=1883 \
-e accessToken=dh1f3NJd86ugeZRB8oML \
-v /etc/localtime:/etc/localtime \
-v mytb-gateway-logs:/thingsboard_gateway/logs \
-v mytb-gateway-extensions:/thingsboard_gateway/extensions \
-v mytb-gateway-config:/thingsboard_gateway/config \
-d --restart always \
thingsboard/tb-gateway

GeXin666, 2025-03-07:
docker run \
--name tb-gateway \
-e host=192.168.80.101 \
-e port=1883 \
-e accessToken=dh1f3NJd86ugeZRB8oML \
-v /etc/localtime:/etc/localtime \
-v mytb-gateway-logs:/thingsboard_gateway/logs \
-v mytb-gateway-extensions:/thingsboard_gateway/extensions \
-v mytb-gateway-config:/thingsboard_gateway/config \
-d --restart always \
thingsboard/tb-gateway

[image]

[image]

imbeacon, 2025-04-07:
Hi Guys,

It was an issue with slave.py file, the sleep period was too low, it is fixed in the master branch and will be available in the next release.