thingsboard/thingsboard-gateway #1683: [HELP] How can i restrict gateway to create devices

vimalgupta, 2025-02-24:
While sending data via gateway, any new device is getting created while pushing the data and i am not finding a way/config settings to control it.

what i am actually looking for is that the gateway only send data for the existing devices, if a device does not exist in things board, it should not create the device and send data.

please let me know if i am missing some config to achieve above.

imbeacon, 2025-03-04:
Hi @vimalgupta,

Thank you for your interest in ThingsBoard IoT Gateway, such as the gateway uses the gateway mqtt API to send data - it cannot control which devices exist and which are not on ThingsBoard. The gateway due to API - can only send connect? to ThingsBoard, and server decided to create a new device or not. Particularly you can control data in rule chains and not save to unwanted devices there.
We will discuss with Product team internally and maybe they will add ability to control do the gateway can or cannot create new devices on ThingsBoard.

Kirinni, 2025-03-13:
I have the same problem.Hope this feature canbe realized.