thingsboard/thingsboard-gateway #1685: [HELP]Can the ThingsBoard Gateway socket not be configured for UDP connections? Why does the listening port show as TCP when I set the type to UDP?

HuaPing2017, 2025-03-04:
[image]

Blastomanie, 2025-03-06:
Hi @HuaPing2017 

I've encountered a similar issue with the socket connector. You need to wrap these four fields with the key "socket", like this :
`"socket": {
"type": "UDP",
"address": "127.0.0.1",
"port": 50000,
"bufferSize": 1024
}
`

imbeacon, 2025-03-11:
Hi @HuaPing2017,

Thank you for your interest in ThingsBoard IoT Gateway.
You are right, there was an issue with old configuration format. Fix is available in related commit. you can use the version from the master branch, or use workaround, suggested by @Blastomanie.
Also, the fix will be available in the next release, it is scheduled on Thursday.

closed 2025-03-20 by imbeacon

HuaPing2017, 2025-04-10:
> Hi [@HuaPing2017](https://github.com/HuaPing2017),
>
> Thank you for your interest in ThingsBoard IoT Gateway. You are right, there was an issue with old configuration format. Fix is available in [...]

Thank you very much, but now I have encountered another problem. When the device client UDP sends a hexadecimal data packet: 88056e00006298000223262001c6e2aff,
1. Starting from hexadecimal 88, FF ends, DAT [0] 88//start flag
DAT [1]//Device Status
DAT [2]//Device current speed high
DAT [3]//Device current speed low
DAT [4]//Current production [4]
DAT [5]//Current production [3]
DAT [6]//Current production [2]
DAT [7]//Current production [1]
DAT [8]//Current total production [4]
DAT [9]//Current total production [3]
DAT [10]//Current total production [2]
DAT [11]//Current total production [1]
DAT [12]//Current meters [4]
DAT [13]//Current meters [3]
DAT [14]//Current meters [2]
DAT [15]//Current meters [1]
DAT [16]//Version numbers: 0 and 1
DAT [17] FF//End
2. The data format received by the device is: starting from hexadecimal 88 and ending with FF
DAT[0] 16#88
DAT [1] setting status
DAT [2] reserved
DAT [3] reserved
DAT [4] reserved
DAT [5] reserved
DAT [6] reserved
DAT[7] 16#FF
3. Current speed of the device - high position in front, low position in back, unit 0.1 meters per minute
4. Current production array - high position first, low position second, units
5. Current total production array - high position first, low position second, units
6. Current meters - high point at the front, low point at the back, in meters
My configuration:
{
"type": "UDP",
"address": "10.16.17.89",
"port": 9989,
"bufferSize": 1024,
"devices": [
{
"address": "10.16.17.89:9988",
"deviceName": "NR21",
"deviceType": "UDP",
"encoding": "utf-8",
"telemetry": [
{
"key": "start_bit",
"byteFrom": 0,
"byteTo": 1,
"type": "hex"
},
{
"key": "machine_status",
"byteFrom": 1,
"byteTo": 2,
"type": "hex"
},
{
"key": "speed",
"byteFrom": 2,
"byteTo": 3,
"type": "int"
},
{
"key": "current_production",
"byteFrom": 3,
"byteTo": 7,
"type": "int"
},
{
"key": "total_production",
"byteFrom": 7,
"byteTo": 11,
"type": "int"
},
{
"key": "current_meters",
"byteFrom": 11,
"byteTo": 15,
"type": "int"
},
{
"key": "end_bit",
"byteFrom": 15,
"byteTo": 16,
"type": "hex"
},
{
"key": "version",
"byteFrom": -2,
"byteTo": -1,
"type": "hex",
"optional": true
},
{
"key": "UdpValue",
"byteFrom": 0,
"byteTo": -1,
"type": "hex"
}
],
"attributes": [
        
],
"attributeRequests": [
       
],
"attributeUpdates": [
     
],
"serverSideRpc": [
       
]
}
],
"logLevel": "INFO",
"name": "YD",
"enableRemoteLogging": false,
"id": "7bfab73b-ce46-4de4-b38e-3a5c0613e623",
"configVersion": ""
}
The telemetry values I received were never what I wanted, and I don't know how to configure or modify them correctly. I need to parse the hexadecimal data and preserve the original device values

HuaPing2017, 2025-04-10:
[image]