thingsboard/thingsboard-gateway #1687: [HELP] Where should I put the response section in the rest.json

Kirinni, 2025-03-07:
**Describe the issue**

**Advanced response** can't work in my **REST Connector**.When I call a REST API via REST Connector,It only return a code without body.  
I tried to put a Yellow Script Rule Node like [tutorial](https://thingsboard.io/docs/iot-gateway/config/rest/#response) and enable the debug mode in the rule chain.The script node couldn't return a true value by   
```return msg.responseExpected;```  

Can anyone tell me how to enable advanced response in REST Connector?

**Configuration** (Attach your configuration file)

[rest.json](https://github.com/user-attachments/files/19117849/rest.json)

**Connector name (If you need help with some connector/converter):**

REST Connector

**Error traceback (If it was raised):**

none

**Versions (please complete the following information):**
 - OS:  Docker
 - Thingsboard IoT Gateway version: 3.7.0

Kirinni, 2025-03-11:
Now I change my config like this:
```json
{
  "host": "0.0.0.0",
  "port": "5000",
  "SSL": false,
  "security": {
    "cert": "~/ssl/cert.pem",
    "key": "~/ssl/key.pem"
  },
  "mapping": [
    {
      "endpoint": "/api/thirdparty/querygoods",
      "response": {
        "responseExpected": true,
        "timeout": 120,
        "responseAttribute": "device_mac"
      },
      "HTTPMethods": [
        "POST",
        "GET"
      ],
      "security": {
        "type": "basic",
        "username": "xmdz",
        "password": "xmdz"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "Device 3",
        "deviceTypeExpression": "tb-REST-Caller",
        "attributes": [
          {
            "type": "string",
            "key": "endpoint",
            "value": "${endpoint}"
          },
          {
            "type": "string",
            "key": "device_name",
            "value": "${device_name}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temp",
            "value": "${temp}"
          }
        ]
      }
    }
  ],
.............
}
```
When I send a requset to this endpoint, the log from tb-gateway container show this:
```
2025-03-11 07:31:54.841 - |INFO| - [rest_connector.py] - rest_connector - __call__ - 646 - CONVERTER CONFIG: {'type': 'json', 'deviceNameExpression': 'Device 3', 'deviceTypeExpression': 'tb-REST-Caller', 'attributes': [{'type': 'string', 'key': 'endpoint', 'value': '${endpoint}'}, {'type': 'string', 'key': 'device_name', 'value': '${device_name}'}], 'timeseries': [{'type': 'double', 'key': 'temp', 'value': '${temp}'}], 'reportStrategy': None}
2025-03-11 07:31:54.962 - |ERROR| - [tb_logger.py] - tb_logger - exception - 185 - Error while post to basic handler: 'Attributes' object has no attribute 'append'
Traceback (most recent call last):
  File "/thingsboard_gateway/connectors/rest/rest_connector.py", line 654, in __call__
    self.modify_data_for_remote_response(converted_data, self.response_expected)
  File "/thingsboard_gateway/connectors/rest/rest_connector.py", line 502, in modify_data_for_remote_response
    data['attributes'].append({'responseExpected': True})
    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Attributes' object has no attribute 'append'
```
Is there has a bug?

imbeacon, 2025-03-11:
Hi @Kirinni,

Thank you for your interest in ThingsBoard IoT Gateway. 
Yes, there was a bug, please find related commit. You can use the version from the master branch or wait for the next release, it is scheduled on Thursday.

closed 2025-03-20 by imbeacon