thingsboard/thingsboard-gateway #1691: Troubleshooting Device Visibility in ThingsBoard with RAK7289 Gateway and MQTT Integration

Skan-der, 2025-03-11:
Dear Community,

I am currently working on integrating a RAK7289 LoRaWAN gateway with ThingsBoard using MQTT for data transmission. While I have made significant progress, I am encountering an issue where the device does not appear in ThingsBoard despite seemingly correct configurations. I would appreciate any insights or guidance to help me resolve this issue.

Here’s a detailed overview of my setup and the problem:

Current Setup

- ThingsBoard Deployment:

Running ThingsBoard in a Docker container.
The ThingsBoard gateway (tb-gateway) is also containerized and configured to connect to the MQTT broker.

- RAK7289 Gateway:

Successfully connected to the ThingsBoard MQTT broker.
Publishing data to the topic: application/1/device/DEVICE_EUI/rx.

- MQTT Payload Structure:

The payload being sent by the RAK7289 gateway is as follows:
`{
    "topic": "application/1/device/DEVICE_EUI/rx",
    "payload": {
        "applicationID": "1",
        "applicationName": "app1",
        "devEUI": "DEVICE_EUI",
        "deviceName": "example_device",
        "timestamp": 1741353903,
        "fCnt": 1,
        "fPort": 1,
        "data": "AQ==",
        "data_encode": "base64",
        "adr": false,
        "rxInfo": [
            {
                "gatewayID": "GATEWAY_EUI",
                "loRaSNR": 7.5,
                "rssi": -74,
                "location": {
                    "latitude": 36.89579,
                    "longitude": 10.18586,
                    "altitude": 21
                },
                "time": "2025-03-07T13:25:03.661349Z"
            }
        ],
        "txInfo": {
            "frequency": 868100000,
            "dr": 0
        }
    }
}`
ThingsBoard Gateway Configuration:

The tb-gateway is configured with the following settings:
`version: '3.4'
services:
  tb-gateway:
    image: thingsboard/tb-gateway
    container_name: tb-gateway
    restart: always
    ports:
        - "5000:5000"
    extra_hosts:
      - "host.docker.internal:host-gateway"
    environment:
      - host=MY_IP #not local host
      - port=1883
      - accessToken=MY_ACCESS_TOKEN
    volumes:
      - tb-gw-config:/thingsboard_gateway/config
      - tb-gw-logs:/thingsboard_gateway/logs
      - tb-gw-extensions:/thingsboard_gateway/extensions
volumes:
  tb-gw-config:
    name: tb-gw-config
  tb-gw-logs:
    name: tb-gw-logs
  tb-gw-extensions:
    name: tb-gw-extensions`

- Data Mapping in ThingsBoard:

The MQTT connector is configured with the following data mapping:

Topic Filter: application/+/device/+/rx
Payload Type: JSON
Device Name: Extracted from ${payload.deviceName}
Profile Name: Set to default
Attributes and Time Series: Configured to parse the payload fields.

[image]

and this is the configuration in request mapping

[image]

Despite the above configurations, the device does not appear in ThingsBoard. I have verified the following:

The RAK7289 gateway is successfully publishing data to the MQTT broker.
The tb-gateway container is running without errors.
The MQTT payload structure matches the expected format.

However, the device is not being created or recognized in ThingsBoard. I suspect there might be an issue with the data mapping or a potential bug in ThingsBoard.
 when following the tutorial  [things board how to add gateway](https://thingsboard.io/docs/iot-gateway/getting-started/?connectorsCreation=mqtt) I succeeded in adding the device demo and receiving values, but I couldn't proceed successfully and don't know where the problem could be.  

Any help would be greatly appreciated.

imbeacon, 2025-03-12:
Hi @Skan-der,

Thank you for your interest in ThingsBoard IoT Gateway.

I have tried the similar payload locally and it is parsed correctly.
Are you sure that your ThingsBoard IoT Gateway MQTT connector connected to the same broker as the RAK gateway? 
I should notice that MQTT connector is used to connect to external brokers and obtain the data from them. Also, regular setup of ThingsBoard doesn’t provide MQTT broker, ThingsBoard has a limited amount of topics that are described in API section of documentation, it doesn’t act like the regular broker.

As far as I know RAK is LoRaWAN gateway, you can easily configure it to connect to some network server and then use integration in ThingsBoard to get the data. There are several type of integrations that will allow you to do this, like Chirpstack,  TheThingsStack, Loriot etc. 

If you want to connect through ThingsBoard IoT Gateway (TB IoT Gateway) - you will need the following setup:

TB IoT Gateway with MQTT connector - connected to some MQTT broker.
RAK7289 - connected to the same MQTT broker.

Also, please check that you are receiving message that you described, because it looks like you may receive a message not with “payload” and “topic”, but just like object in payload: `{“applicationId”: 1…`

To check the payload - you can subscribe to topic and just see what you receive. For example using mosquitto_sub command:

`mosquitto_sub -h YOU_MQTT_BROKER_HOST -t application/+/device/+/rx`

if you are sure that connection is established correctly - please provide example of payload that you receive using command that i wrote.

imbeacon, 2025-04-07:
Closed due to inactivity.

closed 2025-04-07 by imbeacon