thingsboard/thingsboard-gateway #1692: [HELP]Json converter may need to be improved

Kirinni, 2025-03-12:
**Describe the issue**
I have a json msg like this:
```json
{
    "device_mac":"11:11:11:11:11:11",
    "method": "setLED",
    "params": {
        "command":"OFFall",
        "startIndex": 1,
        "ledNum":999,
        "r":50,
        "g":50,
        "b":50
    }
}
```
And converter config is in the below section⬇️.

I tried to converter the **"params"** element to json type from the msg,but the **"params"** elements become a string wrapped in double quotes, and double quotes('\u0022') are replaced with single quotes('\u0027').It just like this:
```
"{\u0027command\u0027: \u0027ON\u0027, \u0027startIndex\u0027: 1, \u0027ledNum\u0027: 3, \u0027r\u0027: 50, \u0027g\u0027: 60, \u0027b\u0027: 20}"
```
When I process the incoming msg,I need to replace '\u0027' to '\u0022' then use JSON.parse() converter string to json.

I hope that elements of json type can not be converted into strings, or keep the original double quotation marks when they are converted into strings.

**Configuration** 
```json
"converter": {
        "type": "json",
        "deviceNameExpression": "Device 3",
        "deviceTypeExpression": "default",
        "attributes": [
          {
            "type": "string",
            "key": "device_mac",
            "value": "${device_mac}"
          },
          {
            "type": "string",
            "key": "method",
            "value": "${method}"
          },
          {
            "type": "json",
            "key": "params",
            "value": "${params}"
          }
        ],
        "timeseries": []
      }
````

**Connector name (If you need help with some connector/converter):**

**REST Connector**

**Error traceback (If it was raised):**

none

**Versions (please complete the following information):**
 - OS: docker
 - Thingsboard IoT Gateway version 3.7.0

VOID-ALIF, 2025-03-12:
Here are three possible solutions to resolve the issue:

Solution 1: Use json.dumps() (if modifying input before processing)
If you have control over the data before sending it to the ThingsBoard Gateway, ensure params is serialized properly:

`import json

params = {
    "command": "OFFall",
    "startIndex": 1,
    "ledNum": 999,
    "r": 50,
    "g": 50,
    "b": 50
}

json_str = json.dumps(params)
print(json_str)`

Solution 2: Change "value": "${params}" to Keep JSON Format
Modify your ThingsBoard converter configuration to avoid treating params as a string:

`{
  "type": "json",
  "key": "params",
  "value": '{"command": "${params.command}", "startIndex": ${params.startIndex}, "ledNum": ${params.ledNum}, "r": ${params.r}, "g": ${params.g}, "b": ${params.b}}'
}`

Solution 3: Use Python Script in ThingsBoard Gateway
If you cannot modify the raw data, add a custom converter script:

`import json

def process_message(msg):
    if "params" in msg:
        # Replace single quotes with double quotes
        msg["params"] = json.loads(msg["params"].replace("\u0027", "\""))
    return msg`

do reach out to me if you need further assistance, thank you

Kirinni, 2025-03-12:
> Here are three possible solutions to resolve the issue:
>
> Solution 1: Use json.dumps() (if modifying input before processing) If you have control over the data before sending it to the ThingsBoard [...]
Thanks for your help😘. But now I can handle the msg correctly,I just hope the json converter can be made easier to use😎.

imbeacon, 2025-03-12:
Hi @Kirinni,

Thank you for your interest in ThingsBoard IoT Gateway.

The correct way to handle payloads like this is using JsonPath expressions, like @VOID-ALIF suggested in his solution 2.

But it is strange behaviour that double quotes was replaced to unicode symbols, we will investigate this point. It is not correct.

Kirinni, 2025-03-12:
Thanks for your help @imbeacon  @VOID-ALIF .🫡
Now I can deal with this situation in many ways, which is a gain.😄

closed 2025-06-03 by imbeacon