thingsboard/thingsboard-gateway #1696: BACnet Version 3.7.1

Isac-enerlab, 2025-03-18:
Good morning, gentlemen

I had already reported a problem reading BACnet devices. Up until version 3.5.1, we had no problems reading telemetry from about 30 devices on the network and their attributes.

However, in the other versions that were released, the service still fails, even though the errors are displayed.

In this latest version, we are finally able to read data, but from a single device.

Once it connects, it ignores all the others, and the tests were done by changing the devices in the queue; it always considers only the first one.

BACnet Connector

 tb-gateway  2025-03-18 14:09:51 - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __process_event - 1175 - Data from Bacnet-1 connector was sent to storage: [ConvertedData(deviceName=SCT_F0_ST001_EM1, deviceType=Store Energy Meter, telemetry=[TelemetryEntry(ts=1742306991534, metadata={}, values={DatapointKey(key=kWh, report_strategy=None): 10.6, DatapointKey(key=kWh1, report_strategy=None): 10.7})], attributes=Attributes(values={}), metadata={})]

{
  "application": {
    "objectName": "TB_gateway",
    "address": "0.0.0.0",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15,
    "deviceDiscoveryTimeoutInSec": 5
  },
  "devices": [
    {
      "deviceInfo": {
        "deviceNameExpression": "SCT_F0_ST001_EM1",
        "deviceProfileExpression": "Store Energy Meter",
        "deviceNameExpressionSource": "expression",
        "deviceProfileExpressionSource": "constant"
      },
      "address": "192.168.XX.210:47808",
      "pollPeriod": 10000,
      "timeseries": [
        {
          "key": "kWh",
          "objectType": "analogValue",
          "objectId": "107",
          "propertyId": "presentValue"
        },
        {
          "key": "kWh1",
          "objectType": "analogValue",
          "objectId": "108",
          "propertyId": "presentValue"
        }
      ]
    },
    {
      "deviceInfo": {
        "deviceNameExpression": "SCT_F0_1_EM1",
        "deviceProfileExpression": "Store Energy Meter",
        "deviceNameExpressionSource": "expression",
        "deviceProfileExpressionSource": "constant"
      },
      "host": "192.168.15.210",
      "port": "47808",
      "deviceId": "1003",
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "temperature",
          "objectType": "analogOutput",
          "objectId": "1",
          "propertyId": "presentValue"
        }
      ],
      "timeseries": [
        {
          "key": "kWh",
          "objectType": "analogValue",
          "objectId": "109",
          "propertyId": "presentValue"
        },
        {
          "key": "kWh1",
          "objectType": "analogValue",
          "objectId": "110",
          "propertyId": "presentValue"
        }
      ]
    }
  ],
  "logLevel": "DEBUG",
  "name": "Bacnet-2",
  "enableRemoteLogging": false,
  "id": "XXXXXXXX-7289-4839-a352-XXXXXXXX",
  "configVersion": "3.7.1"
}

I used the ex. file when we loaded the connector.

**Versions (please complete the following information):**
 - balena Version:           v20.10.43
 - Debian GNU/Linux 12
 - OS/Arch:          linux/arm64
 - Thingsboard IoT Gateway version 3.7.1
 - Python version  3.11.11

Here are the other issues citing the same problem:

https://github.com/thingsboard/thingsboard-gateway/issues/1690

https://github.com/thingsboard/thingsboard-gateway/issues/1620

imbeacon, 2025-03-19:
Hi @Isac-enerlab,

Thank you for your interest in ThingsBoard IoT Gateway.

Could you enable DEBUG logging level for connector and provide logs to us? It will help us to investigate this case.

Isac-enerlab, 2025-03-19:
Good Morning, thank you for your quick response.

Beloow are the logs collected in Balena Docker.

 tb-gateway  2025-03-19 11:43:04 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 207 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 0.0, 'diskUsage': 0.8, 'freeMemory': 7.1, 'freeDisk': '53.4G', 'gwProcessCpuUsage': 2.5, 'gwMemory': 1.4, 'gwProcessMemoryFullInfo': {'rss': 114229248, 'vms': 1655783424, 'shared': 25100288, 'text': 16384, 'lib': 0, 'data': 266960896, 'dirty': 0, 'uss': 113491968, 'pss': 113789952, 'swap': 0}, 'msgsSentToPlatform': 6, 'msgsReceivedFromPlatform': 0}, 'serviceStats': {'storageMsgPulled': 6, 'storageMsgCount': 0, 'platformMsgPushed': 6, 'platformAttrProduced': 0, 'platformTsProduced': 12}, 'connectorsStats': {'Bacnet-2_converter': {'convertersMsgProcessed': 6, 'convertersAttrProduced': 0, 'convertersTsProduced': 12}, 'Bacnet-2': {'storageMsgPushed': 12}}}
 tb-gateway  2025-03-19 11:43:11 - |DEBUG| - [bacnet_uplink_converter.py] - bacnet_uplink_converter - convert - 72 - Converted data: ConvertedData(deviceName=SCT_F0_ST001_EM1, deviceType=Store Energy Meter, telemetry=[TelemetryEntry(ts=1742384591226, metadata={}, values={DatapointKey(key=kWh, report_strategy=None): 10.6, DatapointKey(key=kWh1, report_strategy=None): 10.7})], attributes=Attributes(values={}), metadata={})
 

 tb-gateway  2025-03-19 11:43:21 - |INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 152 - Discovering devices...
 tb-gateway  2025-03-19 11:43:21 - |DEBUG| - [application.py] - application - indication - 32 - (indication) Received APDU: <bacpypes3.apdu.IAmRequest(UnconfirmedRequestPDU) instance at 0x7ffe6af7dfd0>
 tb-gateway  2025-03-19 11:43:21 - |INFO| - [bacnet_connector.py] - bacnet_connector - indication_callback - 121 - Received APDU, from 192.168.15.210, trying to find device...
 tb-gateway  2025-03-19 11:43:21 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - indication_callback - 136 - Device Device(name=SCT_F0_ST001_EM1, address=192.168.xx.210) already added
 tb-gateway  2025-03-19 11:43:21 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 157 - WhoIs request sent to device 192.168.xx.210:47808
 tb-gateway  2025-03-19 11:43:21 - |DEBUG| - [application.py] - application - indication - 32 - (indication) Received APDU: <bacpypes3.apdu.IAmRequest(UnconfirmedRequestPDU) instance at 0x7ffe6af7c6d0>
 tb-gateway  2025-03-19 11:43:21 - |INFO| - [bacnet_connector.py] - bacnet_connector - indication_callback - 121 - Received APDU, from 192.168.15.210, trying to find device...

imbeacon, 2025-03-20:
According to your logs - the gateway receives indication from device with IP 192.168.15.210, but there is no indication from other devices, could you check logs for other IP addresses, we want to figure out does the gateway receives APDU from other devices, except mentioned and how the gateway discovers them, will it discover them or no.

samson0v, 2025-04-17:
Hi @Isac-enerlab, thanks for your interest in ThingsBoard IoT Gateway.
We are happy to tell you that this bug was fixed via https://github.com/thingsboard/thingsboard-gateway/pull/1732
Please, update your Gateway to the newest version via the master branch and let us know about the result.

closed 2025-05-08 by imbeacon