thingsboard/thingsboard-gateway #1697: [BUG] Request Connector does not work for non JSON repsonses

apachler, 2025-03-19:
**Describe the bug**
If using the Request Connector with endpoints that does not not respond with a JSON the connector does not catch this at the time of processing the requests response. I was able to fix this by catching the correct `JSONDecodeError` exception at line https://github.com/thingsboard/thingsboard-gateway/blob/f75981c615728de05ce826533884e5c4630984a2/thingsboard_gateway/connectors/request/request_connector.py#L244

Changing the import
`from json import JSONDecodeError`
to
`from requests.exceptions import JSONDecodeError`
fixes the problem.

**Connector name (If bug in the some connector):**
Request Connector

**Error traceback (If available):**
```
|ERROR| - [request_connector.py] - request_connector - __send_request - 313 - Cannot connect to http://xxx/xxx. Connection error.
|DEBUG| - [request_connector.py] - request_connector - __send_request - 314 - Expecting value: line 1 column 1 (char 0)
```

**Versions (please complete the following information):**
 - OS: SLES 15
 - Thingsboard IoT Gateway version 3.5.1

apachler, 2025-03-19:
It seems the same issues exists in Rest Connector ...

imbeacon, 2025-04-07:
Hi @apachler,

Thank you for your interest in ThingsBoard IoT Gateway and noticing this issue, this fix was added to the master branch and will be available in the next release.

closed 2025-04-10 by imbeacon