thingsboard/thingsboard-gateway #1702: [HELP] Claiming devices using Gateway MQTT API

TodorovS, 2025-03-20:
Dear all, 
I need help understanding/implementing device-side key claiming of devices, connected to TB by gateway with MQTT connector.

My setup:
esp32 <=> HiveMQ cloud broker <=> Thingsboard MQTT Gateway <=> Thingsboard

Communication between esp32 and Thingsboard is verified. Device sends basic JSON payload to topic "sensor/data". 
`{
  "sensorName": "my_test_device",
  "sensorType": "esp32",
  "temp": 25.6,
  "hum": 55
}`

See attached tb_gateway.json and hiveMq.json connector configs below.
Gateway receives payload and successfully provisions new device in Thingsboard with the name "my_test_device" and device profile: esp32. Telemetry values are properly updated with each subsequent message, so the entire MQTT communication chain is working fine, as well as the data mapping of my connector. 

Following the claiming documentation I do a test:
1. Copy the access token of my_test_device from Thingsboard UI.

2. Open MQTTx software and connect **_directly_** to Thingsboard on behalf of my_test_device. I send `{"secretKey":"my_secret_key", "durationMs":60000}` to topic v1/devices/me/claim as per Device MQTT api.
Claiming message is successfully recognized and I claim the the device from user account using the device claiming widget.

3. If I follow the **MQTT Gateway API** "In order to initiate claiming device, send PUBLISH message to the following topic: v1/gateway/claim". 
4. I publish message to v1/gateway/claim to the HiveMQ broker (simulating the device sending it to the gateway). The message JSON payload also from the documentation:
`{
  "my_test_device": {
    "secretKey": "my_secret_key",
    "durationMs": 60000
  }
}`
5. In this setup claiming the device from a user account always fails.
6. The log message from Thingsboard:
2025-03-19 23:48:12,727 [http-nio-0.0.0.0-9090-exec-6] WARN  o.t.s.s.d.ClaimDevicesServiceImpl - Failed to find the device's claiming message![my_test_device].
7. Also the Thingsboard gateway logs do not indicate any message received on v1/gateway/claim topic.

How should my esp32 device properly send the claiming message down the chain HiveMQ => gateway => Thingsboard server so that I can claim the device? What am I doing wrong if I follow the MQTT gateway documentation 1:1?

 - OS: Fedora 40 / docker
 - Thingsboard community edition thingsboard/tb-postgres v.3.9.0
 - Thingsboard IoT Gateway version [latest] thingsboard/tb-gateway 
 - Python version - 3.12.9

[tb_gateway.json](https://github.com/user-attachments/files/19364027/tb_gateway.json)
[hiveMq.json](https://github.com/user-attachments/files/19364028/hiveMq.json)

TodorovS, 2025-03-21:
Okay, it is clear that v1/gateway/claim is the topic where the gateway publishes the incoming claiming requests to the Thingsboard server. 
What I am looking for is:
1. The claiming request payload which device should send to gateway;
2. The topic to which device publishes claiming requests to gateway;
3. The proper data mapping so that gateway extracts device name, secretKey and durationMS from incoming claiming requests.

Is there a built in way to do this, similar to how gateway extracts device name, type and telemetry from this payload:
`{
  "sensorName": "Device_1",
  "sensorType": "thermometer",
  "temp": 111,
  "hum": 1.234
}`
and sends it to v1/gateway/telemetry

Kind regards.

imbeacon, 2025-03-24:
Hi @TodorovS,

Thank you for your interest in ThingsBoard IoT Gateway. 
At the moment, claiming functionality is not added to gateway connectors, but we will add it in the future releases.

icoco, 2025-05-25:
how to set the device profile ? 

topic : v1/gateway/claim

```
{
  "myDevice": {
    "secretKey": "value_A",
    "durationMs": 60000,
    "profile": "my-device-profile" // ???  possible set the device profile here ?
  } 
}
```

icoco, 2025-05-29:
no way to set device profile while use v1/gateway/* ? 

looks rule chain also not support update the device profile ?

imbeacon, 2025-05-29:
Device profile for device can be set on device creation. In case of the gateway you can set deviceType in connect request, it is the same as device profile name.

samson0v, 2026-01-14:
@icoco @TodorovS 

📢 **a little announcement:**
We just launched our [official Discord server](https://discord.com/invite/mJxDjAM3PF), we'd love to see you there!