thingsboard/thingsboard-gateway #1703: Sending Commands from ThingsBoard to RAK3172 to Control a Relay

Skan-der, 2025-03-20:
I have successfully connected my RAK3172 module to the RAK7289 gateway and integrated it with ThingsBoard Community Edition. 

[image]

[image]

Currently, I am able to send data from the MCU to ThingsBoard. Now, I want to implement downlink communication to control a relay connected to the RAK3172 by sending a command from ThingsBoard.

I have added a switch widget to the ThingsBoard dashboard, but I am unsure about the next steps. Should I configure an MQTT integration in the ThingsBoard rule chain, modify the RAK3172 firmware to handle downlink messages, or take another approach? Any guidance on how to properly set up the downlink communication to activate the relay would be greatly appreciated.

[image]

Thank you!

Skan-der, 2025-03-20:
I'm not very sure, but I think this is the path by using rule chain like in the picture .

[image]

I think I can send data to the gateway rak7289, I didn't know actually how to link the switch control to the rule chain and if I'm obliged to use even RPC I just need to the order from the switch to the gateway rak7289

imbeacon, 2025-03-21:
Hi @Skan-der,

You don't actually need to edit default rule chain, just select created device as a device for switch widget. RPC will automatically send through the gateway. you also need to configure RPC section.

By the way, in your rule chain screenshot you put script rule node to RPC Request FROM Device, not "TO" device. It is a different behaviour (client side RPC, when device can send RPC to the server and get response).

If you want to modify the RPC - you need to do this before `rpc call request` node.

Skan-der, 2025-03-21:
I did connect the switch to my device 

[image]
shall I modify these values ?

[image]
my objective is just when switching ON/OFF a relay connect to my MCU rak3172 will turn correspondingly and I thought about MQTT to send the data to the gateway that itself will command the device

Skan-der, 2025-03-21:
After some research, you were absolutely right about the script rule node to RPC connection. I’m currently working on fixing it, and now I’m pretty sure there are four key points I need to accomplish:

Configure the program for switch control

[image]

Create a script to transform and extract data
I still believe we need a script to handle this transformation because it doesn’t make sense to directly connect the MQTT node to the RPC request, right?

[image]

Add and configure the MQTT node
The MQTT node needs to be set up like this:

[image]

Fix the code for the RAK3172 to interpret data from the gateway
Finally, I need to adjust the RAK3172 code so it can properly interpret the data from the gateway.

[image]

Skan-der, 2025-03-24:
Hello Mr @imbeacon 
thanks for the support and actually that MQTT is for the gateway and I even received them in the gateway but than I struggled a bit in the configuration of the code of my device, but I finally managed to find the right path.
My problem was actually in the node transform, as it wasn't sending the right form of Jason file to the gateway.

[image]

[image]

but Mr @imbeacon there is something I didn't understand now everything works fine from TB to gateway to device but in rule chain I'm using

[image]

in the node RPC call request in the debug I'm receiving this

[image]

the state you saw is when I turned on than off the relay from TB

imbeacon, 2025-03-25:
Hi @Skan-der,

Please provide your MQTT connector RPC configuration section. Also, the gateway debug logs will be useful.

closed 2025-04-03 by Skan-der