thingsboard/thingsboard-gateway #1704: [HELP]Does tb-gateway have automatic update feature?

Kirinni, 2025-03-24:
**Describe the issue**

A few days ago, I found that the version was automatically updated to 3.7.2 from 3.7.1.Maybe there is something wrong with my configuration, which makes my gateway unable to work normally, but I don't have time to check it carefully.I have to manually change it back to the old version to work properly.If there has a automatic update feature,how can I disable it?

**Configuration** (Attach your configuration file)
none

**Connector name (If you need help with some connector/converter):**
none

**Error traceback (If it was raised):**
none

**Versions (please complete the following information):**
 - OS: docker
 - Thingsboard IoT Gateway version 3.7.1

imbeacon, 2025-03-25:
Hi @Kirinni,

If you are using "latest" docker image - it is possible, for example when you do "docker compose down" - it removes containers and after next "docker compose up" you may get a new image. 
The gateway also has the following options to be updated:
Docker image was updated by docker, because of using "latest" image in docker-compose.yml file
RPC command - gateway_update
You may update from inside the container just by using pip (I don't think it is this case)

To avoid automatic update you can set image in docker-compose.yml to some specific version - `image: thingsboard/tb-gateway:3.7.1`

Kirinni, 2025-03-26:
Thanks for your help.Now it work well.

closed 2025-03-26 by Kirinni