thingsboard/thingsboard-gateway #1707: [HELP] Use sourceTimestamp for telemetry in OPCUA connector

Spacecast3r, 2025-03-26:
**Issue**
I'm using the ThingsBoard IoT Gateway (version 3.7.2) with the OPC UA connector.
My goal is to use the SourceTimestamp from the OPC UA server instead of the gateway’s system time (datetime.now()) when sending telemetry to ThingsBoard.

I saw something about it [here](https://github.com/thingsboard/thingsboard-gateway/pull/721)

but im not understanding what should be added to the condiguration

Spacecast3r, 2025-03-27:
Ok after reading the source, it seems like the device class was not getting the "timestampLocation" key

```
 def load_values(self):
        self.__configured_values_count = 0
        for section in ('attributes', 'timeseries'):
            for node_config in self.config.get(section, []):
                try:
                    if re.search(r"(ns=\d+;[isgb]=[^}]+)", node_config['value']):
                        child = re.search(r"(ns=\d+;[isgb]=[^}]+)", node_config['value'])
                        self.values[section].append(
                            {'path': child.groups()[0], 'key': node_config['key'],
                             REPORT_STRATEGY_PARAMETER: node_config.get(REPORT_STRATEGY_PARAMETER)})
                    elif re.search(r"\${([A-Za-z.:\\\d\[\]]+)}", node_config['value']):
                        child = re.search(r"\${([A-Za-z.:\\\d\[\]]+)", node_config['value'])
                        self.values[section].append(
                            {'path': self.path + child.groups()[0].split('\\.'), 'key': node_config['key'],
                             REPORT_STRATEGY_PARAMETER: node_config.get(REPORT_STRATEGY_PARAMETER)})

                except KeyError as e:
                    self._log.error('Invalid config for %s (key %s not found)', node_config, e)
            self.__configured_values_count += len(self.values[section])

        self._log.debug('Loaded %r values for %s', len(self.values), self.name)
```

I changed it in:

```
    def load_values(self):
        self.__configured_values_count = 0
        for section in ('attributes', 'timeseries'):
            for node_config in self.config.get(section, []):
                try:
                    if re.search(r"(ns=\d+;[isgb]=[^}]+)", node_config['value']):
                        child = re.search(r"(ns=\d+;[isgb]=[^}]+)", node_config['value'])
                        self.values[section].append(
                            {
                                'path': child.groups()[0],
                                'key': node_config['key'],
                                'timestampLocation': node_config.get('timestampLocation', 'gateway'),
                                REPORT_STRATEGY_PARAMETER: node_config.get(REPORT_STRATEGY_PARAMETER)
                            })
                    elif re.search(r"\${([A-Za-z.:\\\d\[\]]+)}", node_config['value']):
                        child = re.search(r"\${([A-Za-z.:\\\d\[\]]+)}", node_config['value'])
                        self.values[section].append(
                            {
                                'path': self.path + child.groups()[0].split('\\.'),
                                'key': node_config['key'],
                                'timestampLocation': node_config.get('timestampLocation', 'gateway'),
                                REPORT_STRATEGY_PARAMETER: node_config.get(REPORT_STRATEGY_PARAMETER)
                            })
    
                except KeyError as e:
                    self._log.error('Invalid config for %s (key %s not found)', node_config, e)
            self.__configured_values_count += len(self.values[section])
    
        self._log.debug('Loaded %r values for %s', len(self.values), self.name)
```

now its working!

imbeacon, 2025-04-07:
Hi @Spacecast3r,

Thank you for your interest in ThingsBoard IoT Gateway and investigation, I have added your fix to master branch, it will be available in the next release.

closed 2025-04-08 by Spacecast3r