thingsboard/thingsboard-gateway #1710: [HELP]

OYF0910, 2025-03-28:
I'm trying to read device attributes using OPCUA protocol in ThingsBoard Gateway 3.6.3, but I'm getting the following error:  

2025-03-28 14:37:41 - |ERROR| - [opcua_connector.py] - opcua_connector - start_client - 362 - Error in main loop, trying to reconnect: %s  
Traceback (most recent call last):  
  File "C:\Users\other\Desktop\ThingsBoard\thingsboard-gateway-3.6.3\venv\Lib\site-packages\thingsboard_gateway\connectors\opcua\opcua_connector.py", line 332, in start_client  
    await self.__poll_nodes()  
  File "C:\Users\other\Desktop\ThingsBoard\thingsboard-gateway-3.6.3\venv\Lib\site-packages\thingsboard_gateway\connectors\opcua\opcua_connector.py", line 825, in __poll_nodes  
    values = await self.__client.read_attributes(all_nodes)  
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  
  File "C:\Users\other\Desktop\ThingsBoard\thingsboard-gateway-3.6.3\venv\Lib\site-packages\asyncua\client\client.py", line 915, in read_attributes  
    return await self.uaclient.read_attributes(nodeids, attr)  
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  
  File "C:\Users\other\Desktop\ThingsBoard\thingsboard-gateway-3.6.3\venv\Lib\site-packages\asyncua\client\ua_client.py", line 760, in read_attributes  
    data = await self.protocol.send_request(request)  
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  
  File "C:\Users\other\Desktop\ThingsBoard\thingsboard-gateway-3.6.3\venv\Lib\site-packages\asyncua\client\ua_client.py", line 172, in send_request  
    self.check_answer(data, f" in response to {request.__class__.__name__}")  
  File "C:\Users\other\Desktop\ThingsBoard\thingsboard-gateway-3.6.3\venv\Lib\site-packages\asyncua\client\ua_client.py", line 181, in check_answer  
    hdr.ServiceResult.check()  
  File "C:\Users\other\Desktop\ThingsBoard\thingsboard-gateway-3.6.3\venv\Lib\site-packages\asyncua\ua\uatypes.py", line 377, in check  
    raise UaStatusCodeError(self.value)  
asyncua.ua.uaerrors._auto.BadTooManyOperations: The request could not be processed because it specified too many operations.(BadTooManyOperations)  

How can I solve this issue?

OYF0910, 2025-03-28:
I configured 355 keys at once in telemetry properties, and then I got an error. How should I configure it to read all this data at once?

imbeacon, 2025-04-07:
Hi @OYF0910,

Thank you for your interest in ThingsBoard IoT Gateway.
It looks like your OPC-UA server has a limitation for node count to read per time. It looks like a bug, we will fix it, but for now - you can only reduce the amount of time series to read per device. Also you can split your amount of nodes to 4 devices and then in rule chain change device name in metadata before save timeseries rule node to save it correctly

OYF0910, 2025-04-08:
Thank you for your reply. The OPCUA server here is limited to a maximum of 5 connections, and it can currently only read up to 500 attributes. I hope this issue can be resolved quickly. Thank you!

imbeacon, 2025-04-14:
Hi @OYF0910,

We have added functionality to retrieve server limitations, it is available in the master branch and will be available in the next release.

OYF0910, 2025-07-10:
Hello, how should I configure this? Why am I using version 3.7.5 to connect to the OPC UA server, but the ThingsBoard platform can still only collect a maximum of 100 attributes?

OYF0910, 2025-07-12:
ok，thanks！

closed 2025-07-12 by OYF0910