thingsboard/thingsboard-gateway #1711: [HELP]MPU performance requirements of ThingsBoard IoT Gateway

jpiwek, 2025-03-28:
-I am looking into creating a new Secure Gateway platform based on new i.mx91 / i.mx93. What is the requirement on CPU performance, RAM to run ThingsBoard IoT Gateway?
-Additional add-on devices (such as Digital / Analog IO, Sensors, Actuator), will be connected to the gateway over 10BASE-T1S which is a simple switchless Single Pair Ethernet architecture using Multidrop Bus.
-How many request per seconds over Modbus and MQTT can be managed by IoT Gateway?

jpiwek, 2025-07-08:
Thanks for the summary. few questions below.
I am just asking as I am planning a with TB Edge and TB IoT Gateway running on a on RPi CM4. 
3-5 devices will be connected to the Rpi via serial (80 datapoint every 1s-10s) and UDP or TCP (20-30 datapoint every 1s-10s) Connection.
I will think about the message report strategy for my use case.

Q:Are there any plans to switch to Python 3.14? As far as I remember it will support real multi-threading.

Q: My serial protocol is similar to Modbus Serial but support more advanced application features such a datapoint and device discovery. We both already had a thread about it. 

I would like to know how I could optimize my own serial connector to not run into the same performance issue as with Modbus Connector. I plan to start in 2-3 weeks. In my current proprietary serial driver, user selects datapoints of interest and all are read in a single thread sequentially one after another.

samson0v, 2026-01-14:
@jpiwek 

📢 **a little announcement:**
We just launched our [official Discord server](https://discord.com/invite/mJxDjAM3PF), we'd love to see you there!