thingsboard/thingsboard-gateway #1713: [BUG] 3.7.2  Connector for Bacnet

Isac-enerlab, 2025-04-01:
* In this version, there is still a problem with reading only the first device from the JSON.

The test was performed because we are using Docker.

Even creating a second connector, it connects to the first device in the stack and does not maintain a connection with the next one.

**Connector **
BACnet Connector

**Error traceback**

"application": {
    "objectName": "TB_gateway",
    "host": "0.0.0.0",
    "port": "47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceInfo": {
        "deviceNameExpression": "SCT_F0_ST001_EM1",
        "deviceProfileExpression": "Store Energy Meter",
        "deviceNameExpressionSource": "expression",
        "deviceProfileExpressionSource": "constant"
      },
      "host": "192.168.15.210",
      "port": "47808",
      "pollPeriod": 10000,
      "timeseries": [
        {
          "key": "kWh",
          "objectType": "analogValue",
          "objectId": "110",
          "propertyId": "presentValue"
        }

Reading the device stated above

2025-04-01 16:18:34	kWh         10.6

same problem as previous versions.

https://github.com/thingsboard/thingsboard-gateway/issues/1696

**Versions (please complete the following information):**
 - OS: Balena OS
 - Thingsboard IoT Gateway version 3.7.2
 - Python version 3.11.11

imbeacon, 2025-04-09:
Hi @Isac-enerlab,

Thank you for your interest in ThingsBoard IoT Gateway. We have added this issue to our backlog, we are planing to investigate and fix it in release 3.7.4.

samson0v, 2025-04-17:
Hi @Isac-enerlab, thanks for your interest in ThingsBoard IoT Gateway.
We are happy to tell you that this bug was fixed via [PR #1732](https://github.com/thingsboard/thingsboard-gateway/pull/1732)
Please, update your Gateway to the newest version via the master branch and let us know about the result.

closed 2025-05-08 by imbeacon