thingsboard/thingsboard-gateway #1715: Synchronizing Relay States Between Two Devices in Opposite Mode

Skan-der, 2025-04-03:
Hi,

I am currently facing an issue with two devices, each connected to a relay. They send the state of their respective relays:

ON → data = 4

OFF → data = 7

[image]

I want to configure them to work in an opposite synchronized manner, meaning if one device is ON, the other should be OFF.

The initial approach I used:

[image]

I attempted to implement a switch mechanism, but the code did not work as expected:
`// Configuration
var DEVICE1 = "rak3172";
var DEVICE2 = "rak3172_2";
var COMMAND_PORT = 2;

// Only process telemetry
if (msgType !== "POST_TELEMETRY_REQUEST") {
    return { msg: msg, metadata: metadata, msgType: msgType };
}

// Determine device and command
var currentDevice = metadata.deviceName;
var isOnCommand = msg.data === "Bw=="; // "Bw==" = ON, "BA==" = OFF

// Route based on device and state
if (currentDevice === DEVICE1) {
    return {
        msg: {
            target: DEVICE2,
            command: isOnCommand ? "BA==" : "Bw==", // Opposite state
            fPort: COMMAND_PORT,
            confirmed: true
        },
        metadata: metadata,
        msgType: "TO_DEVICE2"
    };
}
else if (currentDevice === DEVICE2) {
    return {
        msg: {
            target: DEVICE1,
            command: isOnCommand ? "BA==" : "Bw==", // Opposite state
            fPort: COMMAND_PORT,
            confirmed: true
        },
        metadata: metadata,
        msgType: "TO_DEVICE1"
    };
}

// For unknown devices
return { msg: null, metadata: metadata, msgType: "NO_ACTION" };`
Is there a better way to approach this problem, or how can I fix the issue?

Skan-der, 2025-04-03:
**_update:_** now it is working fine Whenever I interrupt the system, it fixes itself.

[image]

and this the code of the script that synchronize:
`// Configuration - Set these once
var MASTER_DEVICE = "rak3172";
var CONTROLLED_DEVICE = "rak3172_2";
var COMMAND_PORT = 2;

// Command Mapping - Don't change these
var MASTER_ON = "Bw==";  // 7 (ON from rak3172)
var MASTER_OFF = "BA=="; // 4 (OFF from rak3172)
var CMD_ON = "AQ==";     // 1 (ON to rak3172_2)
var CMD_OFF = "AA==";    // 0 (OFF to rak3172_2)

// Only process messages from the master device
if (msgType === "POST_TELEMETRY_REQUEST" && metadata.deviceName === MASTER_DEVICE) {
    
    // Determine what command to send based on master's state
    var commandToSend = (msg.data === MASTER_ON) ? CMD_OFF : CMD_ON;
    
    // Always send the opposite command immediately
    return {
        msg: {
            device: CONTROLLED_DEVICE,
            fPort: COMMAND_PORT,
            confirmed: true,
            data: commandToSend,
            timestamp: Math.floor(Date.now() / 1000)
        },
        metadata: metadata,
        msgType: "RELAY_CONTROL"
    };
}

// Ignore all other messages
return {
    msg: null,
    metadata: metadata,
    msgType: "IGNORED_MESSAGE"
};`

But I have one major problem: it is sending to the device in a continuous way. I tried to make it send only when the value changed, but I couldn't figure it out. Is there a way to fix this?

[image]

in normal scenarioit should only receive downlink once as the system is stable.

imbeacon, 2025-04-14:
It looks like you sending RPC on each time series update, you need to add some mechanism to avoid this, for example have server attribute that contains required rpc to device and clean it's value when rpc sent, or something like this.

closed 2025-04-14 by imbeacon

Skan-der, 2025-04-14:
yes and now it is fixed now it works fine