thingsboard/thingsboard-gateway #1720: [BUG] Modbus RPC,it reture time out

sgjzwbjk, 2025-04-09:
This is the logs:

2025-04-09 01:34:08 - |INFO| - [modbus_connector.py] - modbus_connector - server_side_rpc_handler - 415 - Received server side rpc request: {'device': 'HW-01-VRF52', 'data': {'id': 49, 'method': 'setPower', 'params': 0}, 'id': 49}
2025-04-09 01:34:08 - |DEBUG| - [modbus_connector.py] - modbus_connector - server_side_rpc_handler - 424 - Modbus connector received rpc request for HW-01-VRF52 with server_rpc_request: {'device': 'HW-01-VRF52', 'data': {'id': 49, 'method': 'setPower', 'params': 0}, 'id': 49}
2025-04-09 01:34:11 - |ERROR| - [modbus_connector.py] - modbus_connector - __read_slave_data - 276 - Timeout error for device HW-03-VRF84 function code 3 address 0, it may be caused by wrong data in server register.
2025-04-09 01:34:11 - |DEBUG| - [bytes_modbus_downlink_converter.py] - bytes_modbus_downlink_converter - convert - 100 - Created builder [0].
2025-04-09 01:34:11 - |DEBUG| - [slave.py] - slave - write - 248 - Write 0 value to address 0 with function code 6
2025-04-09 01:34:11 - |ERROR| - [modbus_connector.py] - modbus_connector - __read_slave_data - 276 - Timeout error for device HW-01-VRF8 function code 3 address 0, it may be caused by wrong data in server register.
2025-04-09 01:34:11 - |DEBUG| - [modbus_connector.py] - modbus_connector - __poll_device - 236 - Polling HW-01-VRF56 slave
2025-04-09 01:34:11 - |DEBUG| - [slave.py] - slave - read - 219 - Reading 10 registers from address 0 with function code 3
2025-04-09 01:34:14 - |ERROR| - [slave.py] - slave - __write - 274 - Failed to write with function code 6: 
2025-04-09 01:34:14 - |DEBUG| - [modbus_connector.py] - modbus_connector - server_side_rpc_handler - 442 - Result: {'response': None}
2025-04-09 01:34:14 - |ERROR| - [modbus_connector.py] - modbus_connector - __read_slave_data - 276 - Timeout error for device HW-01-VRF56 function code 3 address 0, it may be caused by wrong data in server register.
2025-04-09 01:34:14 - |DEBUG| - [modbus_connector.py] - modbus_connector - __poll_device - 236 - Polling HW-01-VRF65 slave
2025-04-09 01:34:14 - |DEBUG| - [slave.py] - slave - read - 219 - Reading 10 registers from address 0 with function code 3
2025-04-09 01:34:14 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_rpc_reply - 1788 - Error while sending RPC reply
Traceback (most recent call last):
  File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 1779, in __send_rpc_reply
    self.tb_client.client.gw_send_rpc_reply(device, req_id, content, quality_of_service=quality_of_service)
  File "/usr/local/lib/python3.11/site-packages/tb_gateway_mqtt.py", line 277, in gw_send_rpc_reply
    info = self._send_device_request(TBSendMethod.PUBLISH, device, topic=GATEWAY_RPC_TOPIC,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tb_gateway_mqtt.py", line 188, in _send_device_request
    info = self._publish_data(**kwargs, device=device_name,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tb_device_mqtt.py", line 997, in _publish_data
    return self._send_request(TBSendMethod.PUBLISH, {"topic": topic, "payload": data, "qos": qos}, timeout,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tb_device_mqtt.py", line 854, in _send_request
    return self.__send_publish_with_limitations(kwargs, timeout, device, msg_rate_limit, dp_rate_limit)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tb_device_mqtt.py", line 910, in __send_publish_with_limitations
    self.__send_split_message(results, part, kwargs, timeout, device, msg_rate_limit, dp_rate_limit, topic)
  File "/usr/local/lib/python3.11/site-packages/tb_device_mqtt.py", line 925, in __send_split_message
    kwargs["payload"] = dumps(part['message'])
                        ^^^^^^^^^^^^^^^^^^^^^^
TypeError: Type is not JSON serializable: _asyncio.Future

I have read the code,
this function "thingsboard_gateway.connectors.modbus.entities.master.Master.write_register"
return this result "Future finished result=False"

imbeacon, 2025-04-11:
Hi @sgjzwbjk,

Thank you for interest in ThingsBoard IoT Gateway,
Could you provide your configuration for RPC that you are using?

imbeacon, 2025-04-11:
Is it:

method - setPower
functionCode - 6
address - 0
objectsCount - 1
?

sgjzwbjk, 2025-04-11:
{
  "master": {
    "slaves": [
      {
        "host": "10.8.18.93",
        "port": 23,
        "method": "socket",
        "unitId": 52,
        "deviceName": "HW-01-VRF52",
        "deviceType": "VRF-Modbus设备配置",
        "timeout": 350000,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "reportStrategy": {
          "type": "ON_REPORT_PERIOD",
          "reportPeriod": 3000
        },
        "type": "tcp",
        "attributes": [],
        "timeseries": [
          {
            "tag": "所有属性",
            "type": "bytes",
            "address": 0,
            "objectsCount": 10,
            "functionCode": 3,
            "reportStrategy": {
              "type": "ON_REPORT_PERIOD",
              "reportPeriod": 15000
            }
          }
        ],
        "attributeUpdates": [],
        "rpc": [
          {
            "tag": "setValue",
            "type": "bits",
            "address": 31,
            "objectsCount": 1,
            "functionCode": 5,
            "bitTargetType": "int"
          },
          {
            "tag": "getValue",
            "type": "bits",
            "address": 31,
            "objectsCount": 1,
            "functionCode": 1,
            "bitTargetType": "int"
          },
          {
            "tag": "setPower",
            "type": "16int",
            "address": 0,
            "objectsCount": 1,
            "functionCode": 6
          }
        ]
      },
      {
        "host": "10.8.18.93",
        "port": 23,
        "method": "socket",
[... 1099 lines cut ...]
            "tag": "setValue",
            "type": "bits",
            "address": 31,
            "objectsCount": 1,
            "functionCode": 5,
            "bitTargetType": "int"
          },
          {
            "tag": "getValue",
            "type": "bits",
            "address": 31,
            "objectsCount": 1,
            "functionCode": 1,
            "bitTargetType": "int"
          }
        ]
      }
    ]
  },
  "name": "hw-15-2-vrf-modbus-01",
  "id": "4baa73a5-cf2d-4369-afd5-df6938971c1c",
  "logLevel": "DEBUG",
  "enableRemoteLogging": true,
  "configVersion": "3.7.2"
}

imbeacon, 2025-04-11:
Could you try to pull the latest image from DockerHub and try with it?

sgjzwbjk, 2025-04-11:
Thanks!
This is the configuration.
I configured 21 devices in a connector configuration When I tried to call the RPC method (setPower), it failed to execute Is it due to an excessive number of devices?
I have tried configuring only 2 devices, and the RPC method can be called normally

imbeacon, 2025-04-11:
Unfortunately, I cannot provide answer for this question, right now, we will try to reproduce and debug, to find the root cause for this issue.

sgjzwbjk, 2025-04-11:
The gateway image version I am currently using is 3.7.2, which should be the latest, right?

imbeacon, 2025-04-11:
No, please try 3.7.3.

imbeacon, 2025-05-21:
Hi @sgjzwbjk,

Looks like the issue was solved in the last release, if no - please open a new issue and add link to this.

closed 2025-05-21 by imbeacon