thingsboard/thingsboard-gateway #1727: Issues with "Update Multiple Attributes" Widget and Rule Chain Time Filter

Skan-der, 2025-04-12:
Hi everyone,

I’ve been working on a project where I need to activate a device based on a specific time interval. To achieve this, I’m using the "Update Multiple Attributes" widget to configure the timing, followed by a rule chain to handle the activation logic. However, I’ve encountered a few issues and would appreciate any guidance.

Problem 1: Shared Attributes vs. Server Attributes
I tested the widget by adding two shared attributes (monday_start_hour and monday_end_hour) to the device.

When I modified these attributes via the widget, the changes were reflected in the server attributes instead of the shared attributes (see screenshots below).

[image]

[image]

[image]

I tried resolving this using widget actions but haven’t had success yet.

Problem 2: Rule Chain Time Filter Script
The second part involves a rule chain that checks whether the current time falls within the configured interval. Here’s the filter script I’m using:

`// 1. Get BOTH attribute types as fallback
var attrs = metadata.serverAttrs || metadata.sharedAttrs || {};
var start = attrs.monday_start_hour;
var end = attrs.monday_end_hour;

// 2. Debug output
metadata.debug = {
  found_attributes: Object.keys(attrs).length > 0,
  start: start,
  end: end,
  current_time: new Date().toString(),
  attribute_source: metadata.serverAttrs ? "server" : 
                   metadata.sharedAttrs ? "shared" : "none"
};

// 3. Validation
if (start === undefined || end === undefined) {
  metadata.debug.error = "Missing time attributes";
  return false;
}

// 4. Time check (Friday forced for testing)
var now = new Date();
now.setDay(5);
var currentHour = now.getHours();

return (currentHour >= start && currentHour < end);
`

Issue:

The script always returns false, seemingly because it fails to retrieve monday_start_hour and monday_end_hour.

The debug logs are empty, making it difficult to diagnose.

Screenshot of Rule Chain Setup:
[image]

Questions:
Why are my widget updates affecting server attributes instead of shared attributes?

How can I ensure the rule chain script correctly reads the time attributes?

Are there any known issues with attribute retrieval in filter scripts?

Any insights or suggestions would be greatly appreciated!

Thanks in advance.

imbeacon, 2025-04-14:
Hi @Skan-der,

It looks like you have a bad widget configuration, probably you chosen server attribute as a data source instead of shared:

[image]

I'm closing this issue, because it is not relate to the gateway itself, please create issues that relates to ThingsBoard itself on https://github.com/thingsboard/thingsboard.

closed 2025-04-14 by imbeacon

Skan-der, 2025-04-14:
thanks for the help and sorry for posting in the wrong place