thingsboard/thingsboard-gateway #1728: Data Display Issue - Dashboard Shows Reverse Values Compared to Device Attributes

Skan-der, 2025-04-12:
Hi everyone,

I’m encountering an issue with data visualization in my ThingsBoard setup. Here’s the scenario:

Current Setup:
Devices send data (e.g., 4 and 7) to a gateway, which forwards it to ThingsBoard.

The device attributes and telemetry show the correct values (4 and 7), as confirmed in the gateway logs and device details.

The Problem:
When displaying this data in a dashboard, the values appear reversed (e.g., the device sending 4 shows 7 and vice versa).

I’ve tried multiple widgets (e.g., cards, gauges), but all display the same incorrect behavior.

Screenshots for Reference:
Gateway and Device Data (Correct):

[image]

[image]

Dashboard Display (Incorrect):

[image]

[image]

Troubleshooting Steps Taken:
Verified that the Save Timeseries node in the rule chain processes the correct values (see screenshot below).

Checked device aliases, widget data sources, and key mappings, but no misconfigurations were found.

Rule Chain Verification:

[image]

[image]

[image]

Questions:
Why is the dashboard showing reversed values despite correct device telemetry?

Could this be caused by a rule chain misconfiguration (e.g., incorrect metadata handling)?

Are there known issues with widget data binding in such scenarios?

Any insights or suggestions would be greatly appreciated!

imbeacon, 2025-04-14:
Hi @Skan-der,

I have tried the same and it worked as expected. Maybe you have set labels or inverted assignment in entity aliases.

You can open issue on ThingsBoard repository and may get the response there, unfortunately we cannot help you with configuring ThingsBoard dashboards.

closed 2025-04-14 by imbeacon

Skan-der, 2025-04-14:
ok thanks for the help mr @imbeacon