thingsboard/thingsboard-gateway #1730: [HELP]rest connector can not successfully run in docker

Liuyuan0803, 2025-04-14:
**Describe the issue**
I deployed thingsboard gateway in Docker, when I use rest connector to communicate with thingsboard, it cannot run successfully.

**Connector name (If you need help with some connector/converter):**
REST Connector

**JSON File**
rest.json:
{
  "host": "10.29.1.1",
  "port": 5000,
  "SSL": false,
  "security": {
    "cert": "~/ssl/cert.pem",
    "key": "~/ssl/key.pem"
  },
  "mapping": [
    {
      "endpoint": "/my_devices",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "${sensorName}",
        "deviceTypeExpression": "${sensorType}",
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "certificateNumber",
            "value": "${certificateNumber}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "endpoint": "/anon1",
      "HTTPMethods": [
        "GET",
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "Device 2",
        "deviceTypeExpression": "default",
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "Model2"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "endpoint": "/anon2",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "custom",
        "deviceNameExpression": "SuperAnonDevice",
        "deviceTypeExpression": "default",
        "extension": "CustomRestUplinkConverter",
        "extension-config": [
          {
            "key": "Totaliser",
            "datatype": "float",
            "fromByte": 0,
            "toByte": 4,
            "byteorder": "big",
            "signed": true,
            "multiplier": 1
          }
        ]
      }
    }
  ],
  "attributeRequests": [
    {
      "endpoint": "/sharedAttributes",
      "type": "shared",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "timeout": 10.0,
      "deviceNameExpression": "${deviceName}",
      "attributeNameExpression": "${attribute}${attribute1}"
    }
  ],
  "attributeUpdates": [
    {
      "HTTPMethod": "POST",
      "SSLVerify": false,
      "httpHeaders": {
        "CONTENT-TYPE": "application/json"
      },
      "security": {
        "type": "anonymous"
      },
      "timeout": 0.5,
      "tries": 3,
      "allowRedirects": true,
      "deviceNameFilter": "SN.*",
      "attributeFilter": ".*",
      "requestUrlExpression": "http://127.0.0.1:5001/",
      "valueExpression": "{\"deviceName\":\"${deviceName}\",\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestUrlExpression": "http://127.0.0.1:5001/${deviceName}",
      "responseTimeout": 1,
      "HTTPMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 10.0,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      },
      "security": {
        "type": "anonymous"
      }
    },
    {
      "deviceNameFilter": "SN.*",
      "methodFilter": "post_attributes",
      "requestUrlExpression": "http://127.0.0.1:5000/my_devices",
      "responseTimeout": 1,
      "HTTPMethod": "POST",
      "valueExpression": "{\"sensorName\":\"${deviceName}\", \"sensorModel\":\"${params.sensorModel}\", \"certificateNumber\":\"${params.certificateNumber}\", \"temp\":\"${params.temp}\", \"hum\":\"${params.hum}\"}",
      "timeout": 10.0,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      },
      "security": {
        "type": "anonymous"
      }
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "HTTPMethod": "POST",
      "valueExpression": "${params}",
      "httpHeaders": {
        "Content-Type": "application/json"
      },
      "security": {
        "type": "anonymous"
      }
    }
  ]
}
tb-gateway.json:
"thingsboard": {
    "host": "10.29.1.1",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "4bff9r7ownvomzx1n5rn"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
    {
     "name": "MQTT Broker Connector",
     "type": "mqtt",
     "configuration": "mqtt.json"
    },
    {
      "name": "REST Connector",
     "type": "rest",
     "configuration": "rest.json"
    }
  ]
}

**Error traceback (If it was raised):**
```
2025-04-14 17:11:46 2025-04-14 09:11:46 - |ERROR| - [tb_logger.py] - tb_logger - exception - 97 - [Errno 99] error while attempting to bind on address ('10.29.3.79', 5000): cannot assign requested address
2025-04-14 17:11:46 Traceback (most recent call last):
2025-04-14 17:11:46   File "/thingsboard_gateway/connectors/rest/rest_connector.py", line 171, in run
2025-04-14 17:11:46     self.__run_server()
2025-04-14 17:11:46   File "/thingsboard_gateway/connectors/rest/rest_connector.py", line 164, in __run_server
2025-04-14 17:11:46     web.run_app(self._app, host=self.__config['host'], port=self.__config['port'], handle_signals=False,
2025-04-14 17:11:46   File "/root/.local/lib/python3.11/site-packages/aiohttp/web.py", line 530, in run_app
2025-04-14 17:11:46     loop.run_until_complete(main_task)
2025-04-14 17:11:46   File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
2025-04-14 17:11:46     return future.result()
2025-04-14 17:11:46            ^^^^^^^^^^^^^^^
2025-04-14 17:11:46 OSError: [Errno 99] error while attempting to bind on address ('10.29.3.79', 5000): cannot assign requested address

```

**Versions (please complete the following information):**
 - OS: Win 10
 - Thingsboard IoT Gateway version [e.g. 3.4.2]

imbeacon, 2025-04-14:
Hi @Liuyuan0803,

Thank you for your interest in ThingsBoard IoT Gateway.
You can set host to 0.0.0.0 and you will be able to receive messages.

Liuyuan0803, 2025-04-14:
ok,I have one more question, should I set port mapping in docker?

imbeacon, 2025-04-14:
Yes, the default port is 5000 and it is mapped in the docker compose file of the gateway by default. If you want to change it - you will also need to change mapping in the docker compose file.

Liuyuan0803, 2025-04-15:
Thank you, it worked

closed 2025-04-15 by Liuyuan0803