thingsboard/thingsboard-gateway #1731: Connectors folder is missing after debian installation!

kashifKhanXeeed, 2025-04-14:
**[Connectors folder is missing](https://thingsboard.io/docs/iot-gateway/install/deb-installation/)**

I followed the above URL to set up the IoT gateway, but the Connectors folder is missing.

Thingsboard version - 3.9.0
OS - Ubuntu 24.04.2 LTS
Gateway version - 3.7.3

imbeacon, 2025-04-15:
Hi @kashifKhanXeeed,

Thank you for your interest in ThingsBoard IoT Gateway.
I have tried deb package from the latest release and everything works as expected, what do you mean "Connectors folder is missing"?
Could you provide the gateway logs to understand what is the issue?

If you are looking for connectors folder of the gateway library - it is located here:
`/var/lib/thingsboard_gateway/venv/lib/python3.12/site-packages/thingsboard_gateway/connectors/`

kashifKhanXeeed, 2025-04-15:
Thanks @imbeacon!

This was helpful, I got the folder.

closed 2025-04-15 by kashifKhanXeeed