thingsboard/thingsboard-gateway #1734: [HELP] Use thingsboard gateway 3.6.3 with Yocto Kirkstone

lagoon-come-strep, 2025-04-14:
**Describe the issue**
On [meta-openembedded](https://github.com/openembedded/meta-openembedded) layer, the recipe for thingsboard-gateway is available.
I'm using Kirkstone and it use an old version of  thingsboard-gateway V2.9 and I want a version closest to the V3.6.2. (I found 3.6.3 available)

When I tried to update the PV and SRC_URI like following I get a lot of missing packages.

```
# Try to change version
PV = "3.6.3"
SRC_URI[sha256sum] = "a5ccad7d6940d194f9f39cc7fc22ab72b3d124066837cde27d92d090ce458a88"
```
I can install it one by one here a list (non exhaustive):
 - orjson
 - psutil
 -  pybase64
 - tb-mqtt-client
 - tb-paho-mqtt-client
 - requests
 - charset_normalizer
 - cachetools
 - grpcio
 - mmh3
 - protobuff
 - pyfiglet questionary
 - service-identity
 - termcolor
 - packaging
 - urllib3
 - ...

but I want to used it from a card without internet access.

**Versions (please complete the following information):**
 - OS: poky like OS
 - Thingsboard IoT Gateway version 3.6.3
 - Python version 3.10.13

imbeacon, 2025-04-15:
Hi @lagoon-come-strep,

Thank you for your interest in ThingsBoard IoT Gateway.
The gateway is designed to install libraries in runtime to avoid installations of unused libraries, but you can install all necessary libraries before running the gateway. The list of this libraries may vary, depending on connectors that you want to use. Usually you may find necessary for connectors libraries in the top of connector file in imports, inside `try ... except ImportError` block.

Here is an example from opcua_connector.py:
```
try:
    import asyncua
except (ImportError, ModuleNotFoundError):
    print("OPC-UA library not found")
    TBUtility.install_package("asyncua")
    import asyncua
```

It installs `asyncua` and import it, the same behavior in other connectors. You can manually install libraries that defined in connectors that you want to use.

lagoon-come-strep, 2025-04-15:
Thank you for your answer, but I realized that I ask my question in the wrong place.
It will still help for later.

All the requirements are listed [here](https://github.com/thingsboard/thingsboard-gateway/blob/master/requirements.txt).

I didn't know about PyPI (where the files from recipe are get), but they also need theses requirements.
e.g. thingsboard_gateway-3.6.2/thingsboard_gateway.egg-info/requires.txt

I expect it to already have it.

As I run it offline, I shall compile from sources instead or install the requirements packages from the host.

closed 2025-04-15 by lagoon-come-strep