thingsboard/thingsboard-gateway #1743: [BUG] Modbus connector service RPC

Jonnik84, 2025-04-18:
**Describe the bug**
RPC to read or write registers seems not working 

[image]

**Connector name (If bug in the some connector):**
Modbus RTU connector

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 22.04LTS]
 - Thingsboard IoT Gateway version [e.g. 3.7.3]
in docker container

imbeacon, 2025-04-18:
Hi @Jonnik84,

Thank you for your interest in ThingsBoard IoT Gateway.
It is requests to connectors, we will investigate this point and fix it.

imbeacon, 2025-06-23:
Hi @Jonnik84,

We are glad to inform that this issue was fixed in https://github.com/thingsboard/thingsboard-gateway/pull/1839, it will be available in the nearest release.

imbeacon, 2025-07-08:
Hi,

The new version is released. Please try to use it, feel free to open new issue if you discover something.

closed 2025-07-08 by imbeacon