thingsboard/thingsboard-gateway #1744: [BUG] "name" field in configuration files can no longer handle space

lagoon-come-strep, 2025-04-18:
**Describe the bug**
In `tb_gateway.json` or `can.json`, set space in the field `name` lead to error. e.g.
```
"connectors": [
    {
      "type": "can",
      "name": "CAN HS I/S",
      "configuration": "canHsIS.json"
    }
  ]
```
rename it CAN_HS_IS fix this problem, but in previous version 3.6.2 I was able to call it with spaces.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
2025-01-08 19:17:23 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __connect_with_connectors - 1063 - ['0d35bd4d-04c1-49a4-9ef2-23529f3d12c9'] Error on loading connector ')
2025-01-08 19:17:23 - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __connect_with_connectors - 1065 - [Errno 2] No such file or directory: 'CAN HS I/S_connector.log.log'
```

**Versions (please complete the following information):**
 - OS: poky OS for raspberry pi (compiled with yocto)
 - Thingsboard IoT Gateway version 3.7.3
 - Python version 3.12.9

imbeacon, 2025-04-24:
Hi @lagoon-come-strep,

The fix is available in the master branch and will be added to the next release of the gateway.

lagoon-come-strep, 2025-04-24:
Ok thank you, then it can be closed. 👍 
Add tests may also help to avoid future regression.