thingsboard/thingsboard-gateway #1748: [HELP] Error on MQTT publication by setting device Assigned firmware

lagoon-come-strep, 2025-04-22:
**Describe the issue**
While trying to create a connector between Thingsboard and SWUpdate to perform OTA, I get an error when setting the field `Assigned firmware` of the gateway in Thingsboard web interface.

It is not possible to use mqtt_firmware_client & thingsboard gateway in the same time, they will be periodicaly disconnected each other.

**Configuration**
No connector configuration is needed the error is handle by setting the value.

**Connector name (If you need help with some connector/converter):**
NA.
But I want to make OTA, so I need access to following subjects:
```
self.subscribe("v1/devices/me/attributes/response/+")
self.subscribe("v1/devices/me/attributes")
self.subscribe("v2/fw/response/+")
```

**Error traceback (If it was raised):**
```
Exception in thread paho-mqtt-client-:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 4528, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 2302, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 1686, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 2105, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 3147, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 3813, in _packet_handle
    return self._handle_publish()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 4153, in _handle_publish
    self._handle_on_message(message)
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 4506, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/usr/lib/python3.12/site-packages/tb_gateway_mqtt.py", line 126, in _on_message
    super()._on_decoded_message(content, message)
  File "/usr/lib/python3.12/site-packages/tb_device_mqtt.py", line 737, in _on_decoded_message
    self.__get_firmware()
  File "/usr/lib/python3.12/site-packages/tb_device_mqtt.py", line 763, in __get_firmware
    self._publish_data(payload, f"v2/fw/request/{self.__firmware_request_id}/chunk/{self.__current_chunk}",
  File "/usr/lib/python3.12/site-packages/tb_device_mqtt.py", line 1154, in _publish_data
    return self._send_request(TBSendMethod.PUBLISH, {"topic": topic, "payload": data, "qos": qos}, timeout,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/tb_device_mqtt.py", line 1012, in _send_request
    return self.__send_publish_with_limitations(kwargs, timeout, device, msg_rate_limit, dp_rate_limit)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/tb_device_mqtt.py", line 1044, in __send_publish_with_limitations
    data = loads(data)
           ^^^^^^^^^^^
orjson.JSONDecodeError: Input is a zero-length, empty document: line 1 column 1 (char 0)
```

**Versions (please complete the following information):**
 - OS: Raspberry pi Custom Linux
 - Last Thingsboard IoT Gateway version 3.7.3
 - Python 3.12.9
 - tb-mqtt-client==1.13.5
 - tb-paho-mqtt-client==2.1.2

lagoon-come-strep, 2025-04-23:
Error was reproduce with docker compose:
ThingsBoard IoT gateway version: 3.7.3
Docker version 27.4.1, build b9d17ea

Inside container version:
Python 3.11.12
tb-mqtt-client==1.13.5
tb-paho-mqtt-client==2.1.2

Raspbian:
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm

imbeacon, 2025-04-23:
Hi @lagoon-come-strep,

Thank you for your interest in ThingsBoard IoT Gateway.
The gateway at the moment doesn't support firmware/software update itself.
According to your situation with several sessions from different client - it is expected behaviour, because by default ThingsBoard  supports only one session per device entity. But you can change it by editing ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE environmental variable and set it to value - more than 1. But it may lead to situation, when you will receive attribute updates or RPC to unexpected client.

According to the error in the gateway - we will check and add handling for this error.

imbeacon, 2025-04-24:
Yes, HTTP - it doesn't create a separated session on the server and may be used beside with running mqtt client without disconnecting issue. Unfortunately, server side rpc - won't give ability to catch firmware link, using default functionality. Because for the gateway - there is no implemented logic in ThingsBoard to deliver firmware/software updates. Only regular device can receive these updates. Whenever, for now - it is possible workaround - to use HTTP client to retrieve firmware for device.

lagoon-come-strep, 2025-04-24:
The issue is fixed, I use [http_firmware_client.py](https://thingsboard.io/docs/user-guide/resources/firmware/http_firmware_client.py) file in parallel.

The tutorial can be found here:
https://thingsboard.io/docs/user-guide/ota-updates/?remoteintegrationdockerinstall=http

It is maybe possible to use OTA system of [LWM2M](https://thingsboard.io/docs/reference/lwm2m-api/) instead (Firmware over-the-air updates section)

As it maybe need reproduction & error handling, I keep the issue open for the moment.