thingsboard/thingsboard-gateway #1749: [HELP]How to deploy TB Gateway on Raspberry Pi?

HuaPing2017, 2025-04-24:
How to deploy TB Gateway on Raspberry Pi? And it is necessary to set the startup to start automatically without using Docker

[e.g. Scoket Connector]

 - OS: [e.g. Raspberry Pi 4B]
 - Thingsboard IoT Gateway version [e.g. 3.8.0]
 - Python version[e.g. 3.7.2]

lagoon-come-strep, 2025-04-24:
Hello HuaPing,
Who can found some information about installation here:
https://thingsboard.io/docs/iot-gateway/install/deb-installation/

Raspberry is based on Debian, you will be able to install package like in Debian linux:
```
lagoon@raspberrypi:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

imbeacon, 2025-04-25:
Hi @HuaPing2017,

If you are installing using debian package - all configuration is located in /etc/thingsboard-gateway.

HuaPing2017, 2025-04-30:
> 您好 ，
> 
> 如果您使用 debian 软件包进行安装 - 所有配置都位于 /etc/thingsboard-gateway/config 中

thanks，it is ok

closed 2025-04-30 by HuaPing2017