thingsboard/thingsboard-gateway #1753: [BUG]

OYF0910, 2025-04-26:
Issue with OPCUA Connector RPC write operation
**Problem Description**
      I'm collecting 3 telemetry data points using OPCUA protocol (2 read-only, 1 read-write). While I can successfully collect all three telemetry values, I encounter an error when trying to modify the read-write attribute using RPC commands.  
**Error Logs：**
2025-04-26 10:54:17 - |ERROR| - [opcua_connector.py] - opcua_connector - server_side_rpc_handler - 1005 - Invalid RPC method name: set

[**RPC Debug Terminal Output**]
Opcua Device-3.7.3＞ set ns=2;s=my.device.rpc_test;1.21
{"result":{"error":"The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)"}}

**Configuration**
"timeseries": [  
  {  
"key": "humidity",  
"path": "${ns=2;s=my.device.湿度}"  
  },  
  {  
"key": "temperature",  
"path": "${ns=2;s=my.device.temperature}"  
  },  
  {  
"key": "rpc_test",  
"path": "${ns=2;s=my.device.rpc_test}"  
  }  
]  
**Environment**
ThingsBoard Gateway version: 3.6.3/3.7.3
Expected Behavior
I expect to be able to write to the read-write variable using RPC commands.

**Questions**
Is the RPC command format correct for writing to an OPCUA variable?
What could be causing the "Invalid RPC method name: set" error?
Why am I getting the "BadWriteNotSupported" error from the OPCUA server?

imbeacon, 2025-04-28:
Hi @OYF0910,

Thank you for your interest in ThingsBoard IoT Gateway!

Could you please provide some additional information about the OPC-UA server you are using? This will help us better understand your setup and solve this issue.

OYF0910, 2025-04-28:
Hi:
I configured it according to the documentation.
  `{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://CAPSSCLT3CXB9Y2:53530/OPCUA/SimulationServer",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "None",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Simulation\\.test",
        "deviceNamePattern": "Simulation Opcua Device",
        "attributes": [],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${ns=3;s=humidity}"
          },
          {
            "key": "temp",
            "path": "${ns=3;s=temp}"
          }
        ],
        "rpc_methods": [

        ],
        "attributes_updates": []
      }
    ]
  },
  "logLevel": "INFO",
  "name": "opcuaSimulation",
  "enableRemoteLogging": false,
  "id": "7fea3152-3481-4d83-b5bc-aaa2018915ec"
}`

Enter the following command in the RPC debug window
**Simulation Opcua Device> set ns=3;s=temp;2**
Then it kept showing the following error.
a)        {"result":{"error":"**The server does not support writing the combination of value, status and timestamps provided.**(BadWriteNotSupported)"}}
a)        Bad_TypeMismatch (0x80740000)，**The value supplied for the attribute is not of the same type as the attribute's value**。

OYF0910, 2025-04-28:
I can successfully write using the uaexpect software, but it doesn't work when using the set method in thingsboard-gateway.
uaexpect：
WriteRequest
  RequestHeader:  RequestHeader
    AuthenticationToken:    b=8wSlyLCrcY6+YOoi+TgzwHuiaihKC3lFDnzWIHdY6Rs=
    Timestamp:    2025-04-26 06:46:10.0447632 GMT
    RequestHandle:    1009880
    ReturnDiagnostics:    0
    AuditEntryId:    null
    TimeoutHint:    10000
    AdditionalHeader:    null
  NodesToWrite:[0]:    WriteValue
      NodeId:      ns=3;s=asd
      AttributeId:      13
      IndexRange:      null
      Value:      DataValue(value=3.2, statusCode=GOOD (0x00000000) "", sourceTimestamp=null, sourcePicoseconds=0, serverTimestamp=null, serverPicoseconds=0)

WriteResponse
  ResponseHeader:  ResponseHeader
    Timestamp:    2025-04-26 06:46:10.0460000 GMT
    RequestHandle:    1009880
    ServiceResult:    GOOD (0x00000000) ""
    ServiceDiagnostics:    null
    StringTable:    null
    AdditionalHeader:    null
  Results:[0]:  DiagnosticInfos:  null

thingsboard-gateway：
WriteRequest
  RequestHeader:  RequestHeader
    AuthenticationToken:    b=oLghS5mXdW+rbH0xxq5ZvqaFYJeydCNHuEwfoex5QGM=
    Timestamp:    2025-04-26 06:46:02.8308830 GMT
    RequestHandle:    1174
    ReturnDiagnostics:    0
    AuditEntryId:    null
    TimeoutHint:    5000
    AdditionalHeader:    null
  NodesToWrite:[0]:    WriteValue
      NodeId:      ns=3;s=asd
      AttributeId:      13
      IndexRange:      null
      Value:      DataValue(value=3.2, statusCode=GOOD (0x00000000) "", sourceTimestamp=2025-04-26 06:46:02.8308830 GMT, sourcePicoseconds=0, serverTimestamp=null, serverPicoseconds=0)
	  
WriteResponse
  ResponseHeader:  ResponseHeader
    Timestamp:    2025-04-26 06:46:02.8350000 GMT
    RequestHandle:    1174
    ServiceResult:    GOOD (0x00000000) ""
    ServiceDiagnostics:    null
    StringTable:    null
    AdditionalHeader:    null
  Results:[0]:    Bad_TypeMismatch (0x80740000) "The value supplied for the attribute is not of the same type as the attribute's value."
  DiagnosticInfos:  null

imbeacon, 2025-04-29:
Thank you for your investigation and report, as we can see - the only one difference is set sourceTimestamp in WriteRequest. I have asked about actually OPCUA server itself, is it ProSYS or something else?

Provided information by you will be enough to solve the issue, but to test the solution it will be good to test on the same environment.

imbeacon, 2025-07-08:
Closed due to inactivity.

closed 2025-07-08 by imbeacon