thingsboard/thingsboard-gateway #1754: Configuring the opcua connector keeps reporting errors

AldirchEugene, 2025-04-28:
|ERROR| - [opcua_connector.py] - opcua_connector - retry_connect_with_backoff - 411 - Encountered error: ValueError('36 is not a valid NodeIdType'). Next connection try in 16 second(s)...

[image]

I configured it according to the official documentation.

[image]

configuration：
{
  "server": {
    "url": "opc.tcp://172.16.8.181:53530/OPCUA/SimulationServer",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 3600000,
    "pollPeriodInMillis": 5000,
    "enableSubscriptions": true,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "deviceNodeSource": "path",
      "deviceNodePattern": "Root\\\\.Objects\\\\.Device1",
      "deviceInfo": {
        "deviceNameExpression": "Device ${Root\\\\.Objects\\\\.Device\\\\.name}",
        "deviceNameExpressionSource": "path",
        "deviceProfileExpressionSource": "constant",
        "deviceProfileExpression": "default"
      },
      "attributes": [
        {
          "key": "humidity",
          "type": "path",
          "value": "${Root\\\\.Objects\\\\.Device1\\.name}"
        }
      ],
      "timeseries": [
        {
          "key": "temperature",
          "type": "path",
          "value": "${Root\\\\.Objects\\\\.Device\\\\.temperature}"
        }
      ],
      "rpc_methods": [],
      "attributes_updates": [],
      "reportStrategy": {
        "type": "ON_REPORT_PERIOD",
        "reportPeriod": 30000
      }
    }
  ],
  "logLevel": "INFO",
  "name": "opcua",
  "enableRemoteLogging": false,
  "id": "e821cf01-7711-44ec-a9c7-728e0b27704f",
  "configVersion": "3.7.3"
}

AldirchEugene, 2025-04-28:
I think your iot-gateway is a failure. It is really hard to use, especially the documentation is not clear.

AldirchEugene, 2025-04-28:
There are a lot of bugs

imbeacon, 2025-04-28:
Hi @AldirchEugene,

Thank you for taking the time to share your experience and feedback.

We’re truly sorry to hear that you found the ThingsBoard Gateway difficult to use and the documentation unclear. Your feedback is very important to us, and we are constantly working to improve both the product and its documentation.

Regarding the specific issue you’re encountering:
	•	Based on the screenshot you shared, it appears that the Root\.Objects\.Device\.name node is missing from your data structure.
	•	The Gateway relies on this path (as configured in your deviceNodePattern) to correctly identify the device name during message processing.
	•	If the expected node does not exist on the server, the Gateway cannot extract the device name, which may lead to  the errors you observed.
	•	From what we can see so far, this issue seems to be related to the structure of your incoming data or the connector configuration, rather than a fault in the Gateway itself.

To help us improve further, could you please clarify:
Which parts of the documentation were unclear for you?
Any specific examples you can provide would be greatly appreciated.

closed 2025-04-30 by imbeacon