thingsboard/thingsboard-gateway #1761: [HELP] pymodbus 3.7.4 works standalone with handle_local_echo=True, but fails inside ThingsBoard Gateway

Vaghulkumar95, 2025-05-02:
**Describe the issue**
Create description about your issue, and your actions to solve it.

We are using the [Waveshare RS232/RS485/CAN board](https://www.waveshare.com/wiki/RS232-RS485-CAN-Board#RS-485_Bus) in RS-485 mode. Since this device echoes Modbus requests, we use handle_local_echo=True.

We tested this with pymodbus 3.7.4 in standalone Python scripts, and it works perfectly (see screenshot below). However, when using ThingsBoard Gateway with the same pymodbus version, the gateway fails to start — even before it attempts any Modbus communication — due to import or enum issues:

{
  "port": "/dev/ttySC0",
  "baudrate": 9600,
  "bytesize": 8,
  "stopbits": 1,
  "parity": "N",
  "timeout": 35,
  "handle_local_echo": true
}

[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[Modbus Connector]

**Error traceback (If it was raised):**
Cannot load GRPC connector!
cannot import name 'Defaults' from 'pymodbus.constants'
Little
Traceback (most recent call last):
  ...
  File "/usr/local/lib/python3.11/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 36, in BytesModbusUplinkConverter
    def from_coils(coils, endian_order=Endian.Little, word_endian_order=Endian.Big):
                                       ^^^^^^^^^^^^^
AttributeError: Little

**Versions (please complete the following information):**
 - OS: [PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian]
 - Thingsboard IoT Gateway version [Version: 3.5.3.1] and also tried in [3.7.3]
 - Python version[3.11.2]
- Pymodbus version [3.0.0] [3.6.1] [3.7.4]

Vaghulkumar95, 2025-05-06:
@samson0v@imbeacon Any updates on this?

Vaghulkumar95, 2025-05-09:
@samson0v @imbeacon Do we have any updates on this? We are going live with this new https://www.waveshare.com/wiki/RS232-RS485-CAN-Board#RS-485_Bus board and need to fix this issue ASAP.

Vaghulkumar95, 2025-05-13:
@samson0v @imbeacon We are going live this week, any help from your end is highly appreciated.

Vaghulkumar95, 2025-05-15:
@samson0v @imbeacon any updates on this?

Vaghulkumar95, 2025-05-16:
@samson0v @imbeacon Is there a way to add "handle_local_echo": true in the modbus.json to avoid echoes?

Vaghulkumar95, 2025-05-19:
Hi @samson0v @imbeacon @ouertani @rroemhild is there a way to use "handle_local_echo": true in modbus.json?

imbeacon, 2025-05-19:
Hi @Vaghulkumar95,

Thank you for your interest in ThingsBoard IoT Gateway.
The gateway uses fixed version of the pymodbus, we have added a task for upgrading the used version of the pymodbus, it will take some time, to upgrade the library, adjust connector to use it and test stability.